cyberbotics/webots #635: Robot fall to pieces during training

xiaoxianSun, 2019-06-24:
**Describe the Bug**
I using Pioneer 3-AT to training a reinforcement algorithm. But during the training process, the robot fall into pieces.   

**Expected behavior**
Is there any operation that i can avoid this problem? Or can I detect this problem through some sensors automaticlly? 

**Screenshots**
[image]

**System**
 - Operating System: Linux Ubuntu 16.04
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

xiaoxianSun, 2019-06-24:
I am using webots2019a

fabienrohrer, 2019-06-24:
How do you reset the robot position?

You can for example reset the entire world like this:

```
from controller import Supervisor

supervisor = Supervisor()
timestep = int(supervisor.getBasicTimeStep())

while supervisor.step(timestep) != -1:
    if supervisor.getTime() > 2.0:  # after 2 seconds
        supervisor.worldReload(). # reload the entire world
```

xiaoxianSun, 2019-06-24:
Thank you for your help! I have try this, but the training speed is slower.I think can combine these two method.
I use the example of projects/languages/python/controllers/driver    translationField.setSFVec3f(self.translation)

fabienrohrer, 2019-06-24:
You should certainly reset the physics just after the `setSFVec3f` call: https://www.cyberbotics.com/doc/reference/supervisor?tab=python#wb_supervisor_simulation_reset_physics

fabienrohrer, 2019-06-24:
> I have try this, but the training speed is slower.I think can combine these two method.

Sorry, I meant using the "supervisor RESET" function which is supposed to be very quick in comparison of the supervisor RELOAD because it reset only the solid positions and the controllers: https://cyberbotics.com/doc/reference/supervisor?tab=python#wb_supervisor_simulation_reset

```python
from controller import Supervisor

supervisor = Supervisor()
timestep = int(supervisor.getBasicTimeStep())

while supervisor.step(timestep) != -1:
    if supervisor.getTime() > 2.0:  # after 2 seconds
        supervisor.worldReset()
```

xiaoxianSun, 2019-06-24:
> 
Thank you ! I will have try this!
> ```

xiaoxianSun, 2019-06-24:
> Sorry, I meant using the "supervisor RESET" function which is supposed to be very quick in comparison of the supervisor RELOAD because it reset only the solid positions and the controllers: [...]

> Thank you! I will have a try!

xiaoxianSun, 2019-06-24:
> Sorry, I meant using the "supervisor RESET" function which is supposed to be very quick in comparison of the supervisor RELOAD because it reset only the solid positions and the controllers: [...]

Thank you ! I will have a try!

fabienrohrer, 2019-06-26:
Did you solved the issue?

fabienrohrer, 2019-06-26:
@xiaoxianSun Did you solved the issue this way?

fabienrohrer, 2019-07-01:
I consider this issue fixed.

@xiaoxianSun Feel free to reopen it if it's not the case 😉

closed 2019-07-01 by fabienrohrer