cyberbotics/webots #649: Windows: Webots crashes if installed in user folder with non-ASCII characters

omichel, 2019-06-25:
If Webots is installed locally (not as administrator) on Windows and the user home path contains non-ASCII characters (Chinese characters for example), then Webots crashes at start-up. However, this is not a regression and I believe many applications have the same problem. Nevertheless, we are probably not far away from a working solution because Webots is actually started, displays the splash screen, main window, then crashes...

DavidMansolino, 2019-11-28:
> Nevertheless, we are probably not far away from a working solution because Webots is actually started, displays the splash screen, main window, then crashes...

This sentence is contradictory with the `wontfix` tag.
In any case, I believe we should either fix it either document it in the known bugs and/or installation instructions and close this isue.

fabienrohrer, 2019-11-28:
By the way, would it be possible to test this in innosetup and do not propose this path in case of problems?

DavidMansolino, 2019-11-28:
That's indeed an interesting solution to investigate.

BenjaminDeleze, 2022-08-29:
@omichel did #5103 fix this?

omichel, 2022-08-29:
Yes, it actually fixed it.

closed 2022-08-29 by omichel