cyberbotics/webots #650: Java Installation Instruction not Working on Ubuntu 16.04

DavidMansolino, 2019-06-25:
**Describe the Bug**
These installation instructions are not working on Ubuntu 16.04 anymore: https://www.cyberbotics.com/doc/guide/using-java#linux

stefaniapedrazzi, 2019-07-01:
The "issue" should affect all the platforms.

The webupd8team's PPA has been discontinued due to changes in the Oracle Java License Agreement that now require to be logged in to download the JDK package.

https://launchpad.net/~webupd8team/+archive/ubuntu/java

Oracle Java SE Licensing FAQ:
https://www.oracle.com/technetwork/java/javase/overview/oracle-jdk-faqs.html

stefaniapedrazzi, 2019-07-01:
Oracle provides open source builds for Windows and macOs but only of JDK >= 12 on https://jdk.java.net

closed 2019-07-01 by stefaniapedrazzi