cyberbotics/webots #666: Failure of the generate_diff.py script

DavidMansolino, 2019-06-27:
**Describe the Bug**
This script is often failing on Travis and AppVeyor, e.g.:
  - https://travis-ci.com/omichel/webots/builds/117098995
  - https://ci.appveyor.com/project/omichel/webots/builds/25578834

With the following stack:
```
if ($env:APPVEYOR_SCHEDULED_BUILD -ne "True" -and $env:APPVEYOR_REPO_TAG -ne "true") { C:\\Python37-x64\\python.exe $env:WEBOTS_HOME\\tests\\sources\\generate_diff.py $env:APPVEYOR_REPO_COMMIT $env:APPVEYOR_REPO_NAME }
C:\\Python37-x64\\python.exe : Traceback (most recent call last):
At line:1 char:88
+ ... e "true") { C:\\Python37-x64\\python.exe $env:WEBOTS_HOME\\tests\\sou ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (Traceback (most recent call last)::String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
  File "C:\projects\webots\\tests\\sources\\generate_diff.py", line 64, in <module>
    url = j["items"][0]["pull_request"]["url"]
IndexError: list index out of range
Command executed with exception:   File "C:\projects\webots\\tests\\sources\\generate_diff.py", line 64, in <module>
    url = j["items"][0]["pull_request"]["url"]
IndexError: list index out of range
```
This perfectly makes sense, we are trying to get the pull-request information for a branch for which there is currently no pull-request (at least for the first commit).

For example on PR #665, I did create and push the two first commits without openning the pull-request => the tests failed, then I open the pull-request and did the third commit => the test passed.

This is not very nice as it shows failled commits (whereas nothing is wrong with them).

fabienrohrer, 2019-06-27:
haha the 666 issue.

I strongly recommend to stop doing git commands in tests. Long tests should be done less often (every night or before a release), that's it.

omichel, 2019-06-27:
Fixed in #670.

closed 2019-06-27 by omichel