cyberbotics/webots #669: Creation of an online real testing on real robots in an artificially created real environment.

EmilEnchev, 2019-06-27:
I would like to have the opportunity to test my Webots controlling software an ONLINE on REAL robots in an artificially created REAL environment against a small fee. I would also love these real robots to include the humanoid robots like NAO. 

Please, ask SoftBank Robotics, do they want to join a similar venture by providing you with a few free test NAO robots.

Best regards,
Emil Enchev

fabienrohrer, 2019-06-27:
Thank you for the report, I will raise this to our weekly meeting next Monday and keep you informed here.

EmilEnchev, 2019-06-27:
Thanks, Fabien

EmilEnchev, 2019-06-27:
To SoftBank:

> "Do you understand, these their robots not good for anything useful, just because they NOT have the appropriate software.  For example, Boston Dynamic does good hardware for robots, but software is the weak link for them. That's why Google is getting rid of them. It was a good business move from Google site. If SoftBank want to make progress they must think outside the box. For example, I will never work for them but I'd love to make paid software for their robots, which their specialists can not. And here we fall into a dilemma. I can make wonderful software for their robots, but I can not test it. RENTING online is simple solution."

fabienrohrer, 2019-07-01:
I just raised the discussion to my colleagues, and we decided we won't go on with your suggestion.

If you have interest in this topic, you could for sure find help on other robotics exchange platforms such as reddit.com.
I wish you good luck with this 👍

closed 2019-07-01 by fabienrohrer