cyberbotics/webots #681: Unable to run matlab example

sxbug, 2019-06-30:
**Describe the Bug**
env：Win10，webots 2019b and MATLAB 2017b. 
Using webots to open its own matlab example, click to run it , the MATLAB window is blocked, there is frozen, can only be  closed by windows' TaskManager .  I have set WEBOTS_HOME, WEBOTS_CONTROLLER_NAME and WEBOTS_VERSION parameters .            
Later, I came up with a solution: empty the content of launcher. m,  then run this example.  After pop up the MATLAB window  stop by error,  I copy the content of launcher. m into the matlab window and run it again. This Webots example can run now, but there are still errors, as if there is some libraries not loaded.

**Steps to Reproduce**
1. open webots2019b
2. load C:\Program Files\Webots\projects\languages\matlab\worlds\e-puck_matlab.wbt
3. run it
4. matlab frozen

**Expected behavior**
Should be the normal operation of the program

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - win10 64bit Chinese version
 - Graphics Card: AMD Radeon R9 M370X

DavidMansolino, 2019-07-01:
This is most probably related to a conflicting library in your `PATH`, please have a look at #63.

sxbug, 2019-07-01:
I had clear windows registry ,and reinstalled matlab, but frozen too.

when I clear launcher.m, and try again. after pause, I  paste origin launcher.m to matlab window 

it can run,and some error msg:
[matlab] Error: �Ҳ���ָ���ĳ�����
[matlab]  (dynamic library)
[matlab] Error: remote control initialisation failed
[matlab] Error: Cannot load the "C:/Program Files/Webots/projects/robots/gctronic/e-puck/plugins/remote_controls/e-puck_bluetooth/e-puck_bluetooth.dll" remote control library.
[matlab] Running Matlab sample Webots controller.

DavidMansolino, 2019-07-02:
This is indeed a known issue caused by conflicting libraries between Matlab and the e-puck remote-control, however, you can safely ignore these errors they will not prevent you from controlling the robot from Matlab.

sxbug, 2019-07-02:
But I can not find a way to ignore these errors, because when I start this simulate ,the matlab is halt and ONLY a black window. What can I do? Thanks

DavidMansolino, 2019-07-02:
Have you tried the other Matlab samples which don't have dedicated temote_control plugins, e.g. youbot_matlab?

sxbug, 2019-07-02:
I had try youbot_matlab, It's the same. Matlab is still stuck.

sxbug, 2019-07-02:
In launcher.m  , I had add two lines:
% set env
setenv('WEBOTS_CONTROLLER_NAME', 'matlab');
setenv('WEBOTS_VERSION', '2019b');

Is that a problem?

DavidMansolino, 2019-07-03:
Probably not, but it should be easy to remove them for testing.

sxbug, 2019-07-03:
[image]
Yes. I want to know how to debug the program and find out what the problem is. pls tell me ,Thanks!

sxbug, 2019-07-03:
INFO: youbot_matlab: Starting controller: "D:\Program Files\MATLAB\R2017b\bin\win64\MATLAB.exe" -nosplash -nodesktop -minimize -r launcher
[youbot_matlab] ����ʹ�� cd
[youbot_matlab] �޷���Ŀ¼�ı�Ϊ C:/PROGRA~1/Webots/projects/robots/kuka/youbot//controllers/matlab (Name is nonexistent or not a directory)��
WARNING: “youbot_matlab”控制器退出，状态：-1。
INFO: youbot_matlab: Starting controller: "D:\Program Files\MATLAB\R2017b\bin\win64\MATLAB.exe" -nosplash -nodesktop -minimize -r launcher
[youbot_matlab] ����ʹ�� cd
[youbot_matlab] �޷���Ŀ¼�ı�Ϊ C:/PROGRA~1/Webots/projects/robots/kuka/youbot/controllers/matlab (Name is nonexistent or not a directory)��
WARNING: “youbot_matlab”控制器退出，状态：-1。
INFO: youbot_matlab: Starting controller: "D:\Program Files\MATLAB\R2017b\bin\win64\MATLAB.exe" -nosplash -nodesktop -minimize -r launcher
[youbot_matlab] ����ʹ�� cd
[youbot_matlab] �޷���Ŀ¼�ı�Ϊ C:/PROGRA~1/Webots/projects/robots/kuka/youbot/controllers/matlab (Name is nonexistent or not a directory)��
WARNING: “youbot_matlab”控制器退出，状态：-1。
INFO: youbot_matlab: Starting controller: "D:\Program Files\MATLAB\R2017b\bin\win64\MATLAB.exe" -nosplash -nodesktop -minimize -r launcher
[youbot_matlab] ����ʹ�� eval
[youbot_matlab] δ���庯�������� 'matlab'��
WARNING: “youbot_matlab”控制器退出，状态：-1。
INFO: youbot_matlab: Starting controller: "D:\Program Files\MATLAB\R2017b\bin\win64\MATLAB.exe" -nosplash -nodesktop -minimize -r launcher
[youbot_matlab] ����ʹ�� eval
[youbot_matlab] δ���庯�������� 'matlab'��
WARNING: “youbot_matlab”控制器退出，状态：-1。

How to solve the Garbled character on the console?            I found that the launcher. m file still had a lot of problems. For example: this KUKA youbot, it couldn't find a matlab controller.
after I change the  cd([WEBOTS_PROJECT 'controllers/youbot_' WEBOTS_CONTROLLER_NAME]);
the matlab  flipped out.
What can I do ？thanks

DavidMansolino, 2019-07-05:
I am still convinced that the problem is because of a conflicting library between matlab and Webots.
Instead of letting Webots start Matlab, can you try using the 'extern' controller and start Martlab and the controller (from within MAtlab) by yourself?
Here is some documentation: https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab=undefined#environment-variables

sxbug, 2019-07-05:
I opened the example program:           
C:\Program Files\Webots\projects\languages\matlab
  Environmental variables are set:  
[image]

sxbug, 2019-07-05:
But MATLAB is still stuck.           
So  I use matlab to opened  the launcher.m , start debugging, found some problems:            
1. I had installed ML_MINGW and configured it with mex-setup. 
Why is it here-------If installed <= 0 | matlab. addons. is Addon Enabled ("ML_MINGW")<== 0            Judgment or not installed?   2.  % initialize libController and redirect stdout/stderr
  try
    calllib('libController', 'wb_robot_init');      %********Run to this matlab flip back  
  catch
    calllib('libController', 'wb_robot_init_msvc');
  end
3. I am sorry for my poor english.
I open the  prog(a copy) and had set controller to <extra>, but I could not start matlab.m too.

Thanks

sxbug, 2019-07-06:
I had downloaded webots 2019a-rev1 again. Without modifying anything, I tried these samples, I found: kuka-youbot can run,  During the running process, the MATLAB window does not display anything.  And after stop matlab display : something is wrong.          
E-puck could not run and MATLAB reported an error.           

 Is there any way to revise it?

Thanks

sxbug, 2019-07-07:
kuka robot error message：

MATLAB crash file:C:\Users\xsun\AppData\Local\Temp\matlab_crash_dump.6548-1:

------------------------------------------------------------------------
             Assertion detected at Sun Jul  7 10:44:55 2019
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled - No sandbox or build area path
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware 
  Default Encoding    : GBK
  Deployed            : false
  Graphics card 1     : Advanced Micro Devices, Inc. ( 0x1002 ) AMD Radeon R9 M370X Version 22.19.170.1536 (2017-7-20)
  Host Name           : MACBOOK-A
  Java Crash Report   : C:\Users\x111sun\AppData\Local\Temp\hs_error_pid6548.log
  MATLAB Architecture : win64
  MATLAB Entitlement ID: 6257193
  MATLAB Root         : D:\Program Files\MATLAB\R2017b
  MATLAB Version      : 9.3.0.713579 (R2017b)
  OpenGL              : hardware
  Operating System    : Microsoft Windows 10 企业版
  Processor ID        : x86 Family 6 Model 70 Stepping 1, GenuineIntel
  Virtual Machine     : Java 1.8.0_121-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : Version 10.0 (Build 17763)

Fault Count: 1

Assertion in void __cdecl `anonymous-namespace'::mwJavaAbort(void) at b:\matlab\src\jmi\jmi\javainit.cpp line 1420:
Fatal Java Exception.  See Java Crash Report for details.

Register State (captured):
  RAX = 0000000010a63338  RBX = 0000000000000000
  RCX = 00000000043edc60  RDX = 0000000000000000
  RSP = 00000000043edbe0  RBP = 000000001e7a9800
  RSI = 0000000010a63338  RDI = 0000000010a63338
 
   R8 = 0000db5afa35bbb7   R9 = 00007ffaaa750000
  R10 = 000000001099b0a8  R11 = 0000000010a53367
  R12 = 00000000043ee2a0  R13 = 0000000010a53268
  R14 = 0000000010a632b8  R15 = 000000001e7a9800
 
  RIP = 000000001085bbaa  EFL = 00000202
 
   CS = 0033   FS = 0053   GS = 002b

Stack Trace (captured):
[  0] 0x0000000010855e83                              bin\win64\libmwfl.dll+00155267
[  1] 0x0000000010854478                              bin\win64\libmwfl.dll+00148600
[  2] 0x00000000108579ce                              bin\win64\libmwfl.dll+00162254
[  3] 0x000000001085727e                              bin\win64\libmwfl.dll+00160382
[  4] 0x000000001085b528                              bin\win64\libmwfl.dll+00177448
[  5] 0x000000001e73df3c                                  bin\win64\jmi.dll+00712508
[  6] 0x00000000659b693d D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02713917
[  7] 0x000000006599eb5b D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02616155
[  8] 0x00000000659b7666 D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02717286
[  9] 0x00000000659bbf6c D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02735980
[ 10] 0x0000000065a523e8 D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+03351528
[ 11] 0x00007ffab5bf477f                      C:\WINDOWS\SYSTEM32\ntdll.dll+00673663
[ 12] 0x00007ffab5b54bef                      C:\WINDOWS\SYSTEM32\ntdll.dll+00019439
[ 13] 0x00007ffab5bf34ee                      C:\WINDOWS\SYSTEM32\ntdll.dll+00668910
[ 14] 0x00000000cfcfe8fe C:\Program Files\Webots\msys64\usr\bin\msys-2.0.dll+00125182
[ 15] 0x00000000cfcffea8 C:\Program Files\Webots\msys64\usr\bin\msys-2.0.dll+00130728
[ 16] 0x00007ffab5bb6b26                      C:\WINDOWS\SYSTEM32\ntdll.dll+00420646
[ 17] 0x00007ffab5b55202                      C:\WINDOWS\SYSTEM32\ntdll.dll+00020994
[ 18] 0x00007ffab5bf34ee                      C:\WINDOWS\SYSTEM32\ntdll.dll+00668910
[ 19] 0x000000003f8140cf                                   <unknown-module>+00000000

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

sxbug, 2019-07-07:
e-puck error message。

MATLAB crash file:C:\Users\x111sun\AppData\Local\Temp\matlab_crash_dump.13924-1:

------------------------------------------------------------------------
             Assertion detected at Sun Jul  7 10:47:47 2019
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled - No sandbox or build area path
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware 
  Default Encoding    : GBK
  Deployed            : false
  Graphics card 1     : Advanced Micro Devices, Inc. ( 0x1002 ) AMD Radeon R9 M370X Version 22.19.170.1536 (2017-7-20)
  Host Name           : MACBOOK-a
  Java Crash Report   : C:\Users\x1111sun\AppData\Local\Temp\hs_error_pid13924.log
  MATLAB Architecture : win64
  MATLAB Entitlement ID: 6257193
  MATLAB Root         : D:\Program Files\MATLAB\R2017b
  MATLAB Version      : 9.3.0.713579 (R2017b)
  OpenGL              : hardware
  Operating System    : Microsoft Windows 10 企业版
  Processor ID        : x86 Family 6 Model 70 Stepping 1, GenuineIntel
  Virtual Machine     : Java 1.8.0_121-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : Version 10.0 (Build 17763)

Fault Count: 1

Assertion in void __cdecl `anonymous-namespace'::mwJavaAbort(void) at b:\matlab\src\jmi\jmi\javainit.cpp line 1420:
Fatal Java Exception.  See Java Crash Report for details.

Register State (captured):
  RAX = 00000000109a3338  RBX = 0000000000000000
  RCX = 00000000e653a1a0  RDX = 0000000000000000
  RSP = 00000000e653a120  RBP = 000000001e789800
  RSI = 00000000109a3338  RDI = 00000000109a3338
 
   R8 = 00000337ce35fc5d   R9 = 00007ffaaa750000
  R10 = 00000000108db0a8  R11 = 0000000010993367
  R12 = 00000000e653a7e0  R13 = 0000000010993268
  R14 = 00000000109a32b8  R15 = 000000001e789800
 
  RIP = 000000001079bbaa  EFL = 00000202
 
   CS = 0033   FS = 0053   GS = 002b

Stack Trace (captured):
[  0] 0x0000000010795e83                              bin\win64\libmwfl.dll+00155267
[  1] 0x0000000010794478                              bin\win64\libmwfl.dll+00148600
[  2] 0x00000000107979ce                              bin\win64\libmwfl.dll+00162254
[  3] 0x000000001079727e                              bin\win64\libmwfl.dll+00160382
[  4] 0x000000001079b528                              bin\win64\libmwfl.dll+00177448
[  5] 0x000000001e71df3c                                  bin\win64\jmi.dll+00712508
[  6] 0x00000000659b693d D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02713917
[  7] 0x000000006599eb5b D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02616155
[  8] 0x00000000659b7666 D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02717286
[  9] 0x00000000659bbf6c D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+02735980
[ 10] 0x0000000065a523e8 D:\Program Files\MATLAB\R2017b\sys\java\jre\win64\jre\bin\server\jvm.dll+03351528
[ 11] 0x00007ffab5bf477f                      C:\WINDOWS\SYSTEM32\ntdll.dll+00673663
[ 12] 0x00007ffab5b54bef                      C:\WINDOWS\SYSTEM32\ntdll.dll+00019439
[ 13] 0x00007ffab5bf34ee                      C:\WINDOWS\SYSTEM32\ntdll.dll+00668910
[ 14] 0x00000000c779e8fe C:\Program Files\Webots\msys64\usr\bin\msys-2.0.dll+00125182
[ 15] 0x00000000c779fea8 C:\Program Files\Webots\msys64\usr\bin\msys-2.0.dll+00130728
[ 16] 0x00007ffab5bb6b26                      C:\WINDOWS\SYSTEM32\ntdll.dll+00420646
[ 17] 0x00007ffab5b55202                      C:\WINDOWS\SYSTEM32\ntdll.dll+00020994
[ 18] 0x00007ffab5bf34ee                      C:\WINDOWS\SYSTEM32\ntdll.dll+00668910
[ 19] 0x000000003f66ec11                                   <unknown-module>+00000000

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

sxbug, 2020-02-17:
Env: webots 2020a rev2, matlab 2017b, 
I loaded C: \ program files \ webots \ projects \ languages \ matlab \ worlds \ e-pump_matlab.wbt file and 
Change the robot's controller to < extern > . then run the launcher. m file(could not run matlab.m file) in Matlab,some error msg displayed :

>> launcher
Warning: name does not exist or is not a directory: C:\Program Files\Webots\lib\matlab 
> In path (line 109)
  In addpath (line 94)
  In launcher (line 23) 
The MATLAB "MinGW-w64 C/C++ Compiler" addon is not installed, please install it from: https://fr.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-compiler
Waiting up to 5 seconds for C:\Users\xsun\AppData\Local\Temp\webots_matlab_lock to be deleted by another MATLAB instance.
Deleting C:\Users\xsun\AppData\Local\Temp\webots_matlab_lock...
Creating: C:\Users\xsun\AppData\Local\Temp\protofile_matlab_2017b_webots_R2020a-rev1.m
Load Library failed.
Error use loadlibrary
Unable to preprocess input file.
The output generated by the preprocessor is:   C:\Program Files\Webots\lib\controller\matlab\allincludes.h:8:61: fatal error: ../../include/controller/c/webots/accelerometer.h: No such file or directory
 #include "../../include/controller/c/webots/accelerometer.h"  
compilation terminated.
Error launcher (line 74)
      loadlibrary( ...

How can I do? Thanks.

sxbug, 2020-02-17:
I have changed /lib/matlab    to    /lib/controller/matlab
But error too. Msg displayed in Matlab:
>> launcher
The MATLAB "MinGW-w64 C/C++ Compiler" addon is not installed, please install it from: https://fr.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-compiler
Waiting up to 5 seconds for C:\Users\xsun\AppData\Local\Temp\webots_matlab_lock to be deleted by another MATLAB instance.
Deleting C:\Users\xsun\AppData\Local\Temp\webots_matlab_lock...
Using prototype file: C:\Users\xsun\AppData\Local\Temp\protofile_matlab_2017b_webots_R2020a_rev2.m
error use loadlibrary
load library "Controller" error
The specified module could not be found.
error  launcher (line 126)
    loadlibrary(libname,protofile,'alias','libController');

Reason:
Wrong use of loaddefinedlibrary
The specified module could not be found.

How can I do? Thanks

omichel, 2020-02-17:
Sorry, we are unable to help you on this until we get a license of MATLAB from the Mathworks. It may happen in the next few weeks.

sxbug, 2020-02-17:
At first Thank you very much for your help.
I want to know if I'm doing the right action:
1. Modified launcher.m and allincludes.h. I think that directory 'Lib\MATLAB' in the webots is wrong. It should be 'lib\controller\MATLAB' , so I modified 2 files.
2. set env :      WEBOTS_CONTROLLER_NAME=matlab
WEBOTS_HOME=C:\Program Files\Webots
WEBOTS_VERSION=R2020a_rev2
Is it right?
3. What other work MUST to been done ?
4. I think whether the loadlibrary function in launcher.m has any fault? 

Thanks!!!

omichel, 2020-02-17:
What you did in 1. seems very correct. In 2. there seems to be a typo, it should be `R2020a-rev2` instead of `R2020a_rev2` and controller name should be `matlab.m` instead of `matlab`. However, I don't know what remain to be done in 3. to repair the MATLAB interface. Can you maybe try to add the path to the Controller.dll in the system PATH environment variable? I am sorry I cannot further help as I don't have MATLAB here.

sxbug, 2020-02-18:
OK, I will try again.
Thanks!!!

sxbug, 2020-02-18:
Crash :(
[image]

DavidMansolino, 2020-02-18:
I did create a new PR to fix all the path issues you mentioned earlier: https://github.com/cyberbotics/webots/pull/1378/files
Please make sure that you have all these modifications locally.

DavidMansolino, 2020-02-19:
We fixed many issues with Matlab in #1378, we have successfully tested it with the latest version of Matlab (R2019b update 4).

closed 2020-02-19 by DavidMansolino

sxbug, 2020-03-11:
There is still a problem with webots connecting to Matlab 2017b. I installed them on another computer（GTX 750 Ti）. The problem is the same, MATLAB is stuck and black. I also tested your method all and installed nightly build version , but It has no effect. I think it is caused by a library that cannot be called. I hope you install a MATLAN 2017b to test it. Thank you!

DavidMansolino, 2020-03-11:
We are fixing an issue with the nightly that was affecting the Matlab interface: #1444
May I ask you to re-try with the nightly of tomorrow?

omichel, 2020-03-11:
Yes, I can reproduce the problem you have with MATLAB 2017b. I have no clue why MATLAB 2017b hangs where MATLAB 2019b works nicely... Unfortunately, fixing support for old versions of MATLAB is not a priority for us.

sxbug, 2020-03-14:
I can use matlab 2017b normally. First, the mingw-64 C / C + + compiler must be installed through the addon functions of MATLAB（ You must register at first）. And then in launcher. m file ,Comment out  line28 -32 ,it is :

%installed_addons = matlab.addons.installedAddons;
%installed = sum(installed_addons.Identifier == "ML_MINGW");
%if installed <= 0 || matlab.addons.isAddonEnabled("ML_MINGW") <= 0
% disp('The MATLAB "MinGW-w64 C/C++ Compiler" addon is not installed, please install it from: https://fr.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-compiler');
%end

So it can be used normally.
I hope it can help you

omichel, 2020-03-16:
I believe I fixed the problem in #1462. Can you confirm?

Eve000017, 2020-03-31:
Hi, I experienced the freeze matlab window as well. 
Then, I try to fixed the problem by #1378 , #1444 , #1462 and I try to follow the method used in #1336. 
I found that I am able to run nao_matlab by using extern (stated in #1336 ).

When I want to try the tutorial 6 (from Webots documentation), I found that the errors occurred. 

[<extern>] Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('calllib')" style="font-weight:bold">calllib</a>
[<extern>] Parameter can not be converted to a character vector
[<extern>] 
[<extern>] Error in <a href="matlab:matlab.internal.language.introspective.errorDocCallback('wb_robot_get_device', 'C:\Users\Desktop\Webots\lib\controller\matlab\wb_robot_get_device.m', 6)" style="font-weight:bold">wb_robot_get_device</a> (<a href="matlab: opentoline('C:\Users\Desktop\Webots\lib\controller\matlab\wb_robot_get_device.m',6,0)">line 6</a>)
[<extern>] result = calllib('libController', 'wb_robot_get_device', name);
[<extern>] 
[<extern>] Error in <a href="matlab:matlab.internal.language.introspective.errorDocCallback('4wheel_IR', 'C:\Users\Desktop\_dir\controllers\4wheel_IR\4wheel_IR.m', 24)" style="font-weight:bold">4wheel_IR</a> (<a href="matlab: opentoline('C:\Users\Desktop\_dir\controllers\4wheel_IR\4wheel_IR.m',24,0)">line 24</a>)
[<extern>]   ds(i) = wb_robot_get_device(ds_names(i));
[<extern>] 
[<extern>] Error in <a href="matlab:matlab.internal.language.introspective.errorDocCallback('launcher', 'C:\Users\Desktop\Webots\lib\controller\matlab\launcher.m', 166)" style="font-weight:bold">launcher</a> (<a href="matlab: opentoline('C:\Users\Desktop\Webots\lib\controller\matlab\launcher.m',166,0)">line 166</a>)
[<extern>]   eval(WEBOTS_CONTROLLER_NAME);

In my opinion, the error occurred is due to the 'Parameter cannot be converted to a character vector'. 
In website (https://www.mathworks.com/matlabcentral/answers/144326-problem-using-external-shared-library-with-pointers), I not really understand how to solve the error. 

Any expert know the way to solve this error?

omichel, 2020-03-31:
The code provided in the tutorial is actually wrong. Moreover the controller cannot be named `4_wheels_collision_avoidance` as MATLAB doesn't accept a file starting with a digit, I would suggest to name it `four_wheels_collision_avoidance` instead. Basically, the error you get can be fixed by using the `convertStringsToChars` function. Here is the correct code:
```MATLAB
TIME_STEP = 64;
ds = [];
ds_names = [ "ds_right", "ds_left" ];
for i = 1:2
  ds(i) = wb_robot_get_device(convertStringsToChars(ds_names(i)));
  wb_distance_sensor_enable(ds(i), TIME_STEP);
end
wheels = [];
wheels_names = [ "wheel1", "wheel2", "wheel3", "wheel4" ];
for i = 1:4
  wheels(i) = wb_robot_get_device(convertStringsToChars(wheels_names(i)));
  wb_motor_set_position(wheels(i), inf);
  wb_motor_set_velocity(wheels(i), 0.0);
end
avoid_obstacle_counter = 0;
while wb_robot_step(TIME_STEP) ~= -1
  left_speed = 1.0;
  right_speed = 1.0;
  if avoid_obstacle_counter > 0
    avoid_obstacle_counter = avoid_obstacle_counter - 1;
    left_speed = 1.0;
    right_speed = -1.0;
  else % read sensors
    for i = 1:2
      if wb_distance_sensor_get_value(ds(i)) < 950.0
        avoid_obstacle_counter = 100;
      end
    end
  end
  wb_motor_set_velocity(wheels(1), left_speed);
  wb_motor_set_velocity(wheels(2), right_speed);
  wb_motor_set_velocity(wheels(3), left_speed);
  wb_motor_set_velocity(wheels(4), right_speed);
  % if your code plots some graphics, it needs to flushed like this:
  drawnow;
end
```
Please let me know if that works for you.
Meanwhile I will fix the tutorial with respect to MATLAB.

Eve000017, 2020-03-31:
Thank you, @omichel 
The code works fine now, The different between the code I used and yours is the missing of 'convertStringsToChars' in my previous code. 

Now, I know the way to run the MATLAB in form of extern. However, it is possible to run it like the C programming in Webots? (without changing the PATH in Environment Variable. )

omichel, 2020-03-31:
> However, it is possible to run it like the C programming in Webots? (without changing the PATH in Environment Variable. )

It is indeed possible to run MATLAB controllers like C controllers. For example I can run the `nao_matlab` this way. Do you get any error message when trying to do so?

Eve000017, 2020-03-31:
> > However, it is possible to run it like the C programming in Webots? (without changing the PATH in Environment Variable. )
> 
> It is indeed possible to run MATLAB controllers like C controllers. For example I can run the `nao_matlab` this way. Do you get any error message when trying to do so?

No error come out, but the environment didnt start even I clicked the Play button. There is also no command display when Play button clicked.

I wonder whether my setting have some issue or not.... That is the reason I use extern to run MATLAB code every time...

omichel, 2020-03-31:
Sometimes, you need to wait a little bit until MATLAB starts. Does it start or not at all? And there is nothing printed in the Webots console?

Eve000017, 2020-03-31:
nothing happen, nothing in the console as well. The MATLAB didnt open as well. 

Do I need to change the PATH or anything?

PR #1462 (2020-03-16): Fix MATLAB launcher
Fixes #681 and #1336.