cyberbotics/webots #702: LED sensor Problem

ExcaliburKong, 2019-07-05:
**Describe the Bug**
Everytime I was trying to add an LED sensor to my robot and edit the LED shape, webots would crash and quit.
**Steps to Reproduce**
1. Go to 'Robot'
2. Click on 'children'
3. Add an 'LED'
2. Click on 'LED's children'
4. See error

**Expected behavior**
Hope I can add the LED sensor.

**Screenshots**
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce GTX 1060  3GB

**Additional context**

omichel, 2019-07-05:
I cannot reproduce this crash. Could you tell me:
- which version of Webots are you using?
- which world file are you using?
- is the simulation running or paused?
- is your robot running some specific controller?
If you are using your own world file, please upload it.

DavidMansolino, 2019-07-16:
Since we can't reproduce the problem and have no feedback I will close this issue. Do not hesitate to give more precisions and we will re-open it.

closed 2019-07-16 by DavidMansolino