cyberbotics/webots #711: Asynchronous setPosition movement

NamelessSuperCoder, 2019-07-11:
So we have a robot that has 6 DOF and we want to make the robot got to specific positions
 smoothly; however, when using the setPosition method it only makes the robot move one joint at a time.

To see this happen:

1. Download this *.zip:
[MinimalisticExample.zip](https://github.com/omichel/webots/files/3383643/MinimalisticExample.zip)
2. Open the MinimalisticExample.wbt
3. Run the simulation

As you can see the robot joints are moving sequentially: one after the other. The next one only moves once the others have reached their targets.

What we would like to see happen is that all the setPosition move at the same time instead. We are aware of setVelocity; however, it would make out life a lot easier if we can just the positions. Also doing setPosition in very small increments like in the ABB example is not what we would like to have to do as that would require us to restructure our code completely.

There are our computer/software specs:
- Windows 10 (latest update)
- Webots R2019b

Many thanks!

DavidMansolino, 2019-07-12:
The motions of the motor are actually done in parallel, but one of them is going way faster than the other.
This is probably because the second motor doesn't have enought torque to move it faster.
I have set the `maxTorque` of both motors to 100 and the `maxVelocity` to 1 (to better see the motion), and here is the result (you can see both joints are moving in parallel).

[image]

You can also see from the position sensors that both are moving at the same time:
[image]

Let me know if anything is unclear.
Please find the modified simulation here: 
[MinimalisticExample.zip](https://github.com/omichel/webots/files/3385991/MinimalisticExample.zip)

NamelessSuperCoder, 2019-07-12:
Thank you very much it seems that this is working! It might be a good idea to put this in the documentation somewhere so that other people know that setPosition might not work if you do not have enough available torque or force.

closed 2019-07-12 by NamelessSuperCoder