cyberbotics/webots #742: supervisor_start_stop_movie test failure

DavidMansolino, 2019-07-23:
The `supervisor_start_stop_movie` test is failing on the `revision` branch (both locally and on travis).

**System**
 - Operating System: Linux Ubuntu 16.04

DavidMansolino, 2019-07-23:
I am now not able to reproduce it anymore, it fails only at the first run.

DavidMansolino, 2019-07-23:
I can't reproduce this on Travis either (I restarted the test and it succeeded).
Let's see if it fails again during one of the nightly.

DavidMansolino, 2019-07-29:
The following test is also sometime failing:
> FAILURE with mirror_supervisor: The white color of the mirror image in the 3D scene is wrong.

https://travis-ci.com/omichel/webots/builds/120988174

fabienrohrer, 2019-11-14:
Until proven otherwise, `start_stop_movie` issues doesn't occur anymore, and `mirror_supervisor` has been fixed here #1083.

closed 2019-11-14 by fabienrohrer