cyberbotics/webots #747: hTc VIVE Pro is not working with We

taffy128s, 2019-07-24:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

DavidMansolino, 2019-09-26:
It seems it is not only the `HTC Vive Pro` headset the problem, but more generally speaking it seems we broke the virtual reality headset interface. As several users reported non-working headset on the latest version and working with Webots R2018b.

omichel, 2021-01-15:
Here is another report from a user:

> I downloaded the oldest version of Webots from the website, which is R2019a version. The VR view worked fine with that version, and it worked file. So, I don't think there was the problem according to the version update of Steam, because the Steam version that used in this experiment was the latest version. I test with 1). Oculus Rift, 2). HTC Vive, and 3). HP Reverb. All devices worked fine.
> 
> However, after 2019a version, the VR version didn't worked, and the results are as bellows :
>
> 1). 2019a-rev1, 2019b-rev1 : I could see just cyan color screen.
> 2). after 2020a-rev1 version : I could see just black screen.
>
> From the results, I speculate there might have some code error between 2019a build and 2019a-rev1 build. And, there might be no relationship between the code error and the update of Steam.

BenjaminDeleze, 2021-11-15:
It is still not working