cyberbotics/webots #752: webots.min.js: rendering issues on some systems

stefaniapedrazzi, 2019-07-25:
On my Linux desktop I get some issues when rendering specific objects, for example:
* `TruckSimple` in the `highway_driving` benchmark
[image]

* both `BungalowStyleHouse` objects in the `humanoid_marathon`
[image]

I tested both on Chrome and Firefox and I get the same issue.
The black dots are visible both the in the robotbenchmark animation and simulation.

@fabienrohrer tested on his macOS Chrome/Firefox and works correctly.

 - Operating System: Linux Ubuntu 18.04
 - Graphics Card: NVIDIA GeForce RTX 2060 6 GB

stefaniapedrazzi, 2019-07-25:
I tried to check if it was a Z-fighting issue and I enabled the logarithmic depth buffer, increased the camera near value and decreased the far value but none of these changes solved the issue.

stefaniapedrazzi, 2019-07-30:
This issue is also visible with the Door object, so I could easily simplify the world:
this is related to IndexedFaceSet (TexturedBox or any simple cube) and the PBRAppearances.normalMap.

stefaniapedrazzi, 2019-07-30:
This seems to be due to texture mapping.
Default IFS texture mapping shows the texture on "front"/"back" faces and the border value is copied along the whole side faces.
I get the issue on the side faces.
But if I set a custom texture mapping where an area (and not a line) of the texture is shown on each face then it works fine without black artifacts.

DavidMansolino, 2019-08-07:
I can reproduce the exact same problem:
```
System: Linux 4.4.0-157-generic x86_64
Processor: Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
Number of cores: 8
OpenGL vendor: NVIDIA Corporation (0x10de)
OpenGL renderer: GeForce GTX 970/PCIe/SSE2 (0x13c2)
OpenGL version: 4.5.0 NVIDIA 384.130
Firefox
```

closed 2019-09-23 by stefaniapedrazzi

stefaniapedrazzi, 2019-09-23:
Just for note I reported the issue in the three.js GitHub (issue  #17559) but they are not interested in fixing it.
Here is the answer:
> Unfortunately seems to be a bug in the driver and/or device. You may want to report the issue to Nvidia instead.

PR #892 (2019-09-23): webots.min.js: texture mapping rendering issue
Fix #752: add some small offset to default texture coordinates to avoid rendering issues if the multiple texture coordinates of a face are exactly the same.

Fix "flat" texture mapping in `TextureParallelepiped` to solve rendering issues with doors in `avoid_obstacles.wbt`.