cyberbotics/webots #754: webots.min.js: bounding box lines affected by post processing

stefaniapedrazzi, 2019-07-25:
The bounding box lines are not shown nicely on the scene in the three.js version because they are affected by the post processing effects.

Instead we should change the rendering order and render this kind of object at the very end so that they look clear.

[image]

stefaniapedrazzi, 2019-07-29:
Given the new roadmap and that this could be a complex development, solving this issue is not a high-priority task.

stefaniapedrazzi, 2019-07-29:
Just note that this doesn't affect only the bounding boxes but all the lines, like the LineSet geometries

Web interface:
[image]

Webots:
[image]

fabienrohrer, 2019-07-29:
I'm wondering if it's not an issue with your GPU, I cannot reproduce the bug:

[image]

Did you tried to disable FXAA? It may alternate line rendering.

stefaniapedrazzi, 2019-07-29:
It is not the FXAA pass that causes issues but the hdrResolvePass.

fabienrohrer, 2019-07-29:
Good clue..

stefaniapedrazzi, 2021-05-26:
To be checked again on R2021b

BenjaminDeleze, 2021-05-26:
fixed in #2769

closed 2021-05-26 by BenjaminDeleze