cyberbotics/webots #757: JOSM Edited map of OSM showing nothing in webots.

nnuk, 2019-07-25:
Hi,

I have edited a OSM map in JOSM but the problem is when i convert that osm to .wbt it is do not opens in the webots application.

I am using Webots 2019b which is the latest version.

Following are the lines which appear in command line while creating .wbt file.

Traceback (most recent call last):
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/importer.py", line 156, in <module>
    parser.parse_file(options.inFile, options.disableMultipolygonBuildings)
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/parser_objects.py", line 52, in parse_file
    self.parse_way(way)
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/parser_objects.py", line 110, in parse_way
    Road.add_to_list(osmId, tags, refs)
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/webots_objects/road.py", line 317, in add_to_list
    if road.parse_tags():
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/webots_objects/road.py", line 117, in parse_tags
    self.forwardLanes = int(self.tags['lanes:forward'])
ValueError: invalid literal for int() with base 10: 'Street'.

Thanks.

fabienrohrer, 2019-07-25:
Sometimes, OSM data contains weird information. Normally a `lanes:forward` should contain only integers, but here, someone introduced a String containing "Street" instead :-) which is certainly a rare mistake.

I just created this patch to support dummy values for `lanes:forward`: https://github.com/omichel/webots/pull/759

Could you give a try? (Simply replacing `WEBOTS_HOME/resources/osm_importer/webots_objects/road.py` by [this file](https://github.com/omichel/webots/blob/improve-osm-importer/resources/osm_importer/webots_objects/road.py))

nnuk, 2019-07-25:
@fabienrohrer Thanks alot for your help now I am getting the map in Webots. But In the command line I am getting these lines :

Invalid "turn:lanes:backward" tag for ""
Invalid "turn:lanes:forward" tag for ""
Invalid "turn:lanes:backward" tag for ""
Invalid "turn:lanes:forward" tag for ""
Invalid "turn:lanes:forward" tag for ""
Invalid "turn:lanes:forward" tag for ""
Invalid "turn:lanes:backward" tag for ""
Invalid "turn:lanes:forward" tag for ""
Invalid "turn:lanes:backward" tag for ""
Invalid "turn:lanes:forward" tag for ""
Invalid "turn:lanes:backward" tag for ""
Invalid "turn:lanes:backward" tag for ""

Traceback (most recent call last):
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/importer.py", line 188, in <module>
    Area.export(outputFile, options.noParkings)
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/webots_objects/area.py", line 203, in export
    treesFile = area.generate_tree_file(os.path.dirname(os.path.abspath(file.name)))
  File "/Volumes/Intenso/Webots Application/WebotsB/Webots.app/resources/osm_importer/webots_objects/area.py", line 255, in generate_tree_file
    os.remove('forest/' + str(self.OSMID) + '.forest')
OSError: [Errno 2] No such file or directory: 'forest/100355349.forest'

Can you kindly guide me through it.?

Thanks.

fabienrohrer, 2019-07-25:
I just updated https://github.com/omichel/webots/pull/759 with a patch which should hopefully solve your issue. This time, it involves another file: area.py to be copied at the right location... Is it working better with it?

fabienrohrer, 2019-07-25:
(The warning about "invalid "turn:lanes:backward" tag" could certainly be ignored. There are other weirdness in your OSM files about lane tags.)

nnuk, 2019-07-25:
@fabienrohrer  Thankyou very much for your help. It worked.

closed 2019-07-25 by nnuk

fabienrohrer, 2019-07-25:
Thank you. The patch will be released in the next Webots version.

omichel, 2019-07-27:
It is indeed already released in the latest [nightly build](https://github.com/omichel/webots/releases)