cyberbotics/webots #783: No simulation showing on first start after changing graphics card on Mac OS High Sierra with Radeon RX 560

WolfgangFahl, 2019-08-06:
**Describe the Bug**
Middle part of screen stays gray / empty on first start of Webots taking guided tour.

**Steps to Reproduce**
1. Download -rw-r--r--@ 1 wf  staff  1278188115 Aug  4 10:24 webots-R2019b.dmg
2. Install
3. Start Webots with old graphics card ATI Radeon 5770 1 GB VRAM
4. Get a complaint that card has not enough ram
5. Stop computer
6. Install Radeon RX 560 4 GB VRAM
7. Restart computer
8. Restart Webots
9. Start guided tour
10. Middle part stays gray while controls seem to run simulation

**Expected behavior**
Should show simulation.

**Screenshots**
N/A since on second start of application simulation works.

**System**
 - Operating System: [Mac OS 10.13.6 High Sierra]
 - Graphics Card: [Radeon RX 560]

**Additional context**
Mac Pro Mid 2010, 12 core, 64 GB RAM

fabienrohrer, 2019-08-06:
Thank you for your bug report.

If I understand well, you can now start Webots smoothly. Only the first startup leads to a crash.

- If it's the case, a new GPU installation may have trick the Webots startup settings causing a crash. We won't investigate this issue further as this is an uncommon case.
- If this is not the case, could you try:

    1. To run Webots in [safe mode](https://cyberbotics.com/doc/guide/starting-webots#on-linux-and-macos). This will set the GPU settings at minimum, and open Webots on an empty world:

            export WEBOTS_SAFE_MODE=true
            cd /Applications/Webots
            ./webots

    2. To determine if Webots detects successfully your freshly installed GPU by running the Webots sysinfo:

            cd /Applications/Webots
            ./webots --sysinfo

    3. To determine if other applications using OpenGL are working.

WolfgangFahl, 2019-08-06:
@fabienrohrer  - thx for the swift reply.

The software did not crash. It simply didn't show anything in the middle of the screen. Another Open GL app might haven been EPIC Games launcher which I am going to deactivate for future tests.
```

./webots --sysinfo
System: Darwin 17.7.0 x86_64
Processor: Intel(R) Xeon(R) CPU           X5690  @ 3.47GHz
Number of cores: 24
OpenGL vendor: ATI Technologies Inc.
OpenGL renderer: AMD Radeon RX 560 OpenGL Engine
OpenGL version: 2.1 ATI-1.68.22
```

fabienrohrer, 2019-08-06:
Ok, I seems worth to test the [safe mode](https://cyberbotics.com/doc/guide/starting-webots#on-linux-and-macos) mentioned above:

    export WEBOTS_SAFE_MODE=true
    cd /Applications/Webots
    ./webots

WolfgangFahl, 2019-08-06:
I'll try it for #784

WolfgangFahl, 2019-08-23:
see #784

fabienrohrer, 2019-11-14:
I explain this bug by the GPU switch. Reinstalling cleanly the OS is certainly the solution.

closed 2019-11-14 by fabienrohrer