cyberbotics/webots #784: animated_skin.wbt crashes on High Sierra 10.13.6

WolfgangFahl, 2019-08-06:
**Describe the Bug**
Trying to run animated_skin.wbt (e.g. as part of guided tour) crashes Webot application.

**Steps to Reproduce**
1. Start Webots guided tour 
2. Click next until animated_skin.wbt appears and tries to run

Alternatively load the world directly and press run button.

**Expected behavior**
No Crash.

**Screenshots**
[image]

**System**
 - Operating System: Mac OS 10.13.6
 - Graphics Card: [Radeon RX 560.]

**Additional context**
Crash report was sent to support@cyberbotics.com

fabienrohrer, 2019-08-06:
I just give a try: I downloaded a fresh R2019b release, and I can run animated_skin.wbt smoothly:

[image]

I unfortunately bet again on some weirdness with your GPU.
Changing a GPU on a mac is not recommended. Are you sure that your system recognize it correctly? Is there some manipulation to do in such case?

WolfgangFahl, 2019-08-23:
The GPU is just fine `Radeon RX 560:

  Chipsatz-Modell:	Radeon RX 560
  Typ:	GPU
  Bus:	PCIe
  Steckplatz:	Slot-1
  PCIe-Lane-Breite:	x8
  VRAM (dynamisch, maximal):	4096 MB
  Hersteller:	AMD (0x1002)
  Geräte-ID:	0x67ff
  Versions-ID:	0x00cf
  Metal:	Unterstützt, Funktionsset macOS GPUFamily1 v3`

WolfgangFahl, 2019-08-23:
Tried the safemode today. Still crashes.

fabienrohrer, 2019-08-26:
Thank you for your tests. We actively test Webots on several Mac with different GPUs and we still cannot reproduce your issue.

The only option I see to further investigate, is to ask you to provide us a stack of the crash. The OS generally opens a dialog box after a crash, containing the stack.

fabienrohrer, 2019-11-14:
I explain this bug by the GPU switch. Reinstalling cleanly the OS is certainly the solution.

closed 2019-11-14 by fabienrohrer

WolfgangFahl, 2019-12-04:
Please reopen the bug. Why wouldn't you want to support this environment?

fabienrohrer, 2019-12-04:
Could you confirm you modified the hardware of your computer?