cyberbotics/webots #791: Linux installation instructions are broken

omichel, 2019-08-09:
**Describe the Bug**
I tried to follow the Linux installation instructions on a fresh Ubuntu 18.04 VM and it fails badly.

**Steps to Reproduce**
1. Follow the instructions from the user guide to install Webots from apt.
2. When I typed: `sudo apt-get install webots` I got:
```bash
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package webots
```

DavidMansolino, 2019-08-09:
I can reproduce this on a fresh native Ubuntu 18.04 too :disappointed:

omichel, 2019-08-09:
When downloading the `.deb` file and opening it with the default application for `.deb` file, it displays "License: proprietary" whereas it should display Apache 2.0 / open source instead. Moreover, a screenshot is missing...
[image]

omichel, 2019-08-09:
Partially fixed in #792.

omichel, 2019-08-09:
It remains the missing screenshot and wrong license to fix.

omichel, 2019-08-13:
It is not possible to have a screenshot until we get officially accepted in the main ubuntu repos: See
https://askubuntu.com/questions/117130/how-to-include-a-screenshot-of-my-application-in-the-deb-ppa-to-show-it-in-softw

omichel, 2019-08-14:
It is not possible to display a free open source license in the ubuntu software center if our package is not part of the official ubuntu repositories in `main` or `universe` sections, see: https://stackoverflow.com/questions/18017084/copyright-file-in-deb-package-how-to

omichel, 2019-08-14:
I started the procedure to get Webots into the official debian repositories. This is the procedure to follow to get it in ubuntu repositories as well. I just found a [debian sponsor](https://mentors.debian.net/sponsors) who will help us getting things done.

omichel, 2019-08-15:
I am closing this issue as the resolution of it implies getting Webots into Ubuntu universe which is described in #810.

closed 2019-08-15 by omichel