cyberbotics/webots #794: webrots crash in Windows7

lj1065, 2019-08-10:
Every time, i run it, when it in "Finalizing nodes", it will crash, and stop working. The environment is Windows7. I  have update the GPU Driver.

lj1065, 2019-08-10:
The error report is "webots.exe 中的 0x0000000180037fa4 处有未经处理的异常: 0xC0000005: 读取位置 0x0000000000000000 时发生访问冲突", in English is, "An unhandled exception occurred at 0x0000000180037fa4 in webots.exe: 0xC0000005: Access violation occurred while reading location 0x0000000000000000"

DavidMansolino, 2019-08-12:
Is this occurring with every world or just one in particular?

Can you please give us the output of the `webots --sysinfo` command?

Can you please try to launch Webots in safe mode: https://cyberbotics.com/doc/guide/starting-webots#safe-mode

lj1065, 2019-08-12:
No, it crash in the first start and every time. 
I follow your suggestiones, i launch it in safe mod, it crashed and it tell me that "webots requires OpenGL3.3 while only OpenGL 1.1 can be initialized,. please check your CPU abilites and install the latest graphics drives ...."
When I quit the safe-mode, and set the WEBOTS_SAFE_MODE to TRUE in the Environment Variables.
It does not crash anymore.
My computer has the latest AMD drive, but i do not know why it crash and why it goes well.

Thanks for your reply.

omichel, 2019-08-12:
It seems you are missing an appropriate driver for your AMD card. It should provide an OpenGL version much better than 1.1. Can you report your exact GPU hardware and OpenGL driver version?

lj1065, 2019-08-13:
CPU Type : Intel(R) Core(TM) i5-4210U CPU @ 1.70GHz
Windows Version : Windows 7 (Service Pack 1) (64 bit)
System Memory : 12 GB
Radeon Software Version : 19.7.5
Radeon Software Edition : Adrenalin 2019

lj1065, 2019-08-13:
AMD Radeon R9 M275X
The version of OpenGL : 3.3.13570 Core Profile Froward-Compatible Context 0

omichel, 2019-08-13:
Sorry, I am running out of ideas about this problem. Maybe it's a driver bug... Can you try Webots on a different machine or on the same machine with Linux?

lj1065, 2019-08-13:
Yeah, I do not know why and the problem is solved after I  set the WEBOTS_SAFE_MODE to TRUE in the Environment Variables. It does not crash anymore. It runs well in my computer now.
Thanks for your reply.

omichel, 2019-08-13:
The safe mode is simply disabling all advanced 3D rendering techniques. Can you try to re-enable 3D features one-by-one (from the Webots / Tools / Preferences / OpenGL) to identify which one is causing the crash?

DavidMansolino, 2019-08-27:
You will find the procedure to start Webots in safe mode here: https://cyberbotics.com/doc/guide/starting-webots#safe-mode

lj1065, 2019-08-27:
Yes, I am glad to do that. But what should i do for testing it. Please tell me.
PS : My computer has two graphics cards. The reason for the problem may be that I have not switched the right graphics card.

fabienrohrer, 2019-11-14:
No feedback from the user since 3 months, not seen in other systems.

closed 2019-11-14 by fabienrohrer

venkateshbotta, 2020-08-09:
same problem for me also. Eventhough i changed environmental variables it is not working.please someone help me.

DavidMansolino, 2020-08-10:
@venkateshbotta which version of Webots are you using? What do you get when you start Webots with the `--sysinfo` argument?