cyberbotics/webots #832: Spinning beach ball of death on Macos when using robot.simulationReset()

WolfgangFahl, 2019-08-23:
**Describe the Bug**
A clear and concise description of what the bug is.
In the inverted_pendulum benchmark example on a Desktop Webots I set E-puck supervisor to TRUE and use robot.simulationReset() in my code

**Steps to Reproduce**
1. Use the example code
2. modify it 
```
@@ -1,10 +1,12 @@
 """Sample Webots controller for the inverted pendulum benchmark."""
 
 from controller import Robot
+from controller import Supervisor
 import math
 
 # Get pointer to the robot.
-robot = Robot()
+#robot = Robot()
+robot=Supervisor()

robot.simulationReset()
```

**Expected behavior**
No Mac OS beachball of death.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - MacOS 10.3.6
 - Graphics Card: MSI Aero ITX Radeon RX 560

**Additional context**
Add any other context about the problem here.

WolfgangFahl, 2019-08-23:
Workaround: save things. Restart Webots. Be happy ...

DavidMansolino, 2019-08-23:
Have you tried using ``worldReload`` instead of ``simulationReset``?

DavidMansolino, 2019-08-26:
I have checked both with the `worldReload` and `simulationReset` functions and both seem to work fine (on Ubuntu).
@fabienrohrer can you try on mac?

fabienrohrer, 2019-08-26:
Thank you for this bug report, I can indeed reproduce it.

A workaround is to save the simulation at step 0 after modifying the E-puck.supervisor field and reload the simulation. Probably the node is in a bad state when changing the field from the API.

fabienrohrer, 2019-08-26:
The infinite loop is because of this test:
https://github.com/omichel/webots/blob/revision/src/webots/nodes/WbRobot.cpp#L742 :

`((WbRobot *) this)->mSupervisor.value()` is correctly equals to `true` at this stage.
The controller is started when expected. 
The `supervisor` flag is well sent to the libController.

DavidMansolino, 2020-07-14:
I can't reproduce the problem in the future R2020b release (develop branch) on macOS.

closed 2020-07-14 by DavidMansolino

WolfgangFahl, 2020-07-14:
I am looking forward to the new release ...