cyberbotics/webots #840: generate_thumbnails.py script is unusable on Windows

omichel, 2019-08-29:
This script has several problems:

1. all the md files are rewritten with Windows line ending causing a huge diff.
2. all the jpeg thumbnails are regenerated (even if not needed)
3. some md files point to the wrong images (with no thumbnail).

Fixing the line ending is easy:

```
import io

def search_and_replace(filename, fromString, toString):
    """Search and replace string in a file."""
    with io.open(filename, 'r') as file:
        data = file.read()
    data = data.replace(fromString, toString)
    with io.open(filename, 'w', newline='\n') as file:
        file.write(data)
```

But that's only one out of 3 problems 😒...

DavidMansolino, 2019-08-29:
They are some slight image differences when the image are generated on Windows (probably due to a different jpeg compression):

[image]

fabienrohrer, 2019-08-29:
> some md files point to the wrong images (with no thumbnail).

Could you mention examples of this?

DavidMansolino, 2019-11-18:
Once the issue with different images generated on different platforms we should restore the thumbnail test: b6f9a7dcdb23afc90194f929f140c7a1862df28a