cyberbotics/webots #844: Improve Python and Java Robot::setMode function

DavidMansolino, 2019-08-30:
**Describe the Bug**

The ``Robot::setMode`` function takes as argument a void pointer, this does not translate fine in Python and Java, and is almost unusable.
We should override this function in the Python and Java wrappers to take another type of argument (such as a string for example).

See #843

stefaniapedrazzi, 2019-12-04:
Using a string argument for Python and Java, and `void *` for C and C++ could cause issues if the remote controller library expects an array of a type different than an array of characters.
So we cannot get rid of the current `setMode` functions, but we can add some new ones for Python and Java that will work only if the remote control libraries expects an array of characters (more common case).

On the other hand, I checked our examples of remote control and all of them are sending arguments with `char *` format.
So another option is to change the C function from `wb_robot_set_mode(int, void*)` to `wb_robot_set_mode(int, char*)`.

fabienrohrer, 2019-12-04:
> So another option is to change the C function from wb_robot_set_mode(int, void*) to wb_robot_set_mode(int, char*).

Naively, I would go for this solution. It is simpler and it allows a better communication in case where the plugin language does not match the controller language.

stefaniapedrazzi, 2019-12-05:
Ok, I also think changing the C function is the better solution.

stefaniapedrazzi, 2019-12-10:
Fixed in #1160

closed 2019-12-10 by stefaniapedrazzi