cyberbotics/webots #858: make failed at the tag 'R2019b'

uniqueluo, 2019-09-04:
**Description**
I'm getting the following linking error when building webots 'R2019b' on Ubuntu 16.04 

- #
- # * dependencies *
- #
- #
- # * ode *
- #
- #
- # * glad *
- #
- #
- # * wren *
- release echo #
- # compiling Renderable.cpp
- Renderable.cpp: In member function ‘void wren::Renderable::recomputeBoundingSphereInViewSpace(const mat4&)’:
- Renderable.cpp:96:42: error: no match for ‘operator=’ (operand types are ‘glm::vec3 {aka glm::tvec3<float, (glm::precision)0u>}’ and ‘glm::tmat4x4<float, (glm::precision)0u>::col_type {aka glm::tvec4<float, (glm::precision)0u>}’)
-        mBoundingSphereInViewSpace.mCenter = viewMatrix * glm::vec4(mBoundingSphere.mCenter, 1.0f);
- 

**Steps to Reproduce**
1. git colne https://github.com/cyberbotics/webots.git
2. cd webots
3. make
4. sudo apt-get install ibglm-dev
5.make

**Expected behavior**
make sucessful

**System**
 - Operating System: Linux Ubuntu 16.04
 - Graphics Card: [ NVIDIA GeForce GTX 106 6 GB.]

DavidMansolino, 2019-09-04:
It seems the ``glm`` submodule is not initialized.
Can you please run the following commands in your 'webots' directory:
```
git submodule init
git submodule update
```

and then re-run the make.

uniqueluo, 2019-09-05:
Fellow your instruction, everything works fine,  thank you!

DavidMansolino, 2019-09-05:
You're welcome

closed 2019-09-05 by DavidMansolino