cyberbotics/webots #859: The temp folder is not cleared properly

omichel, 2019-09-04:
Webots uses a temporary folder to store several files while running.
When Webots quits normally, it deletes its own temporary folder.
However, if it crashes (or the the system crashes or is rebooted), this temporary folder is not deleted.
So, many temporary folders can therefore accumulate and saturate the disk overt time.
To work around this, we implemented this quick and dirty trick: in the geter of the temp folder, we check if only one instance of Webots is running. If this is the case, then we delete all the webots temp folders except for the current one. This is not good because in case we are on a simulation server, it is very unlikely that only one instance of Webots is running at a time. Therefore the temp folder will never get cleaned-up.

A better approach would be the following: when we start Webots, we check all the webots temp folder and we delete the ones which are older than one hour. We should also add a timer in Webots that will touch its own temp folder every hour to prevent another instance of Webots from deleting it.

Such a mechanism has two advantages on the previous one:

1. It removes the need to determine if only one instance of Webots is running. We achieved this by a complicated non cross-platform approach which causes confinement issues with snap.
2. It will be more efficient on simulation servers where the cleanup could occur even if several instances of Webots are running.

Please let me know what do you think about this proposal and if you agree, I will implement it.

fabienrohrer, 2019-09-04:
That sounds good to me.

omichel, 2019-09-05:
OK, thank you for the feedback. I will implement this in #855 (as it is needed to fix yet another snap strict confinement problem).

fabienrohrer, 2019-09-05:
It would be much cleaner to create a PR for this. #855 is already difficult to review because it touches lot of unrelated things.

PR #863 (2019-09-09): Improved life cycle of the Webots temporary folder
Fixes #859.
closed 2019-09-09 by omichel