cyberbotics/webots #873: Webots Help menu refactor

fabienrohrer, 2019-09-10:
The Webots help menu does not reflect anymore our public services state.

[image]

Proposal to refactor the 3 last items:

```
GitHub repository...  # => https://github.com/cyberbotics/webots
Cyberbotics website...  # => https://www.cyberbotics.com
Report an issue...   # => https://github.com/cyberbotics/webots/issues/new/choose
Follow us >
- Subscribe to the Newsletter... # => https://www.cyberbotics.com/news/subscribe.php
- Join the Discord chanel... # => https://discordapp.com/invite/nTWbN9m
- Twitter... # => https://twitter.com/webots
- youTube... # => http://www.youtube.com/user/cyberboticswebots 
```

DavidMansolino, 2019-09-10:
We should indeed have done this a while ago !

fabienrohrer, 2019-09-10:
And the `Linked In` link..

omichel, 2019-09-10:
I would keep the link to the support ticket as it is nice to the users who paid for a Premier service and it may encourage more people to purchase a Premier service.

fabienrohrer, 2019-09-10:
Yes, it seems better. Maybe I could add a `(Premier Service)` suffix in order to warn the regular users they have to login.

fabienrohrer, 2019-09-10:
Fixed in #874

closed 2019-09-10 by fabienrohrer