cyberbotics/webots #891: Wrong Rendering if TextureCoordinate are Equal

DavidMansolino, 2019-09-17:
**Describe the Bug**
If the texture coordinate of the 3 vertices of a triangle are equal the rendering is not homogeneous as expected but looks completely wrong.

**Screenshots**
[image]

**System**
 - Operating System:Linux Ubuntu 18.04
 - Graphics Card: NVIDIA GeForce GTX 970 

**Additional context**
This is reproducible with the following world:
```
#VRML_SIM R2019b utf8
WorldInfo {
}
Viewpoint {
  orientation -0.5610947722608096 -0.7525912045976002 -0.3446434901516687 1.2782008704957197
  position -0.23236190965843073 1.1022376123073423 0.9162881547270028
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Shape {
  appearance CementTiles {
  }
  geometry IndexedFaceSet {
    coord Coordinate {
      point [
        0 0 0
        1 0 0
        1 0 1
      ]
    }
    texCoord TextureCoordinate {
      point [
        0 0
        0 0
        0 0
      ]
    }
    ccw FALSE
    coordIndex [
      0, 1, 2, -1
    ]
    normalIndex [
      0, 1, 2, -1
    ]
  }
}
```

DavidMansolino, 2019-09-17:
The problem is visible only if using a normal map texture.

stefaniapedrazzi, 2019-09-18:
This issue is also visible with the Laptop PROTO

Webots screenshot:
[image]

stefaniapedrazzi, 2019-09-18:
I will check if I can identify where does the issue come from.

stefaniapedrazzi, 2019-09-18:
The problematic function is `cotangentFrame` in `pbr.frag`:
https://github.com/cyberbotics/webots/blob/bd8e1aff86dac1802ad42c2690e19fca50f6375f/resources/wren/shaders/pbr.frag#

For example if I change the `invmax` value to 1.0, the issue disappears.
But I'm still debugging to understand where is exactly the source of the problem and how to prevent it.

PR #895 (2019-09-19): Improve robustness of cotangentFrame function in pbr shader
Fix #891: the case where all the texture coordinates are equal (i.e. `dFdx(uv) == vec2(0) && dFdy(uv) == vec2(0)`) was not handled in `cotangentFrame` function leading to a division by 0.
closed 2019-09-19 by stefaniapedrazzi