cyberbotics/webots #893: Drop Old Qt robot windows

fabienrohrer, 2019-09-18:
Old robot windows based on Qt are causing more and more issues.

- It encourages an obsolete usage.
- It costs maintenance effort due to a complex maintenance of 2 systems.
- It prohibits to have a clean library path space for the controller because it forces the use of webots library path (containing Qt) in every controller. This is particularly visible since #879.
- on macOS, a warning is present in these windows because qt.conf is not found (and can only be added in the controllers launching them, such as every Darwin-OP controller and `void` :-( ):

        [obstacle] objc[73953]: Class QMacAutoReleasePoolTracker is implemented in both /Users/fabien/repos/webots/Contents/Frameworks/QtCore.framework/Versions/5/QtCore (0x101a8c3f8) and /Users/fabien/software/Qt5.11.2/5.11.2/clang_64/lib/QtCore.framework/Versions/5/QtCore (0x10957f3f8). One of the two will be used. Which one is undefined.

Remaining robot windows are: old generic windows (no more used) and qt_utils, darwin-op window (containing code to cross-compile the robot :-( ), BotStudio, OSM interface and GUI Tracker sample.
Darwin-OP windows is certainly the most problematic one.

fabienrohrer, 2019-09-18:
I would vote to simply drop all of this :-) Is this reasonable?

omichel, 2019-09-18:
Do you mean dropping the Darwin-op, BotStudio, OSM interface and GUI tracker robot windows? I believe we could drop the BotStudio one, but not the others.

DavidMansolino, 2019-09-18:
We should at least as a first step migrate the simple 'qt-based' robot windows (such as the automobile one).

fabienrohrer, 2019-09-18:
I forgot the vehicle window (description updated).

fabienrohrer, 2019-09-18:
Yes, I really mean to drop them all. This is a bit extreme, but I think it's a brake to go further. HTML windows is the way to go for our next objectives, we should focus on this, not on past things.

- BotStudio: should be replaced by Blockly or so, when we are ready for this. We have no urgency there.
- OSM window: we can use the OSM importer using the script in a much more efficient way. Every project I worked is using the script.
- vehicle window: it shows few tabs that are in theory simple to port to the generic HTML window (speed, steering angles, brakes, encoders, throttles, etc.). This is the only window where I would do a porting effort. 
- GUI tracker: is already advantageously replaced by the HTML windows sample.  
- DARwIn-OP: this is the most arguable part. But should we bother by a 7 years old robot? Couldn't we do the same as for the Nao?
- old generic window and qt_utils are currently there for backward compatibility and to let work the vehicle window and the Darwin-op tabs.

fabienrohrer, 2019-09-18:
But maybe it's worth to discuss of this in the next alignment meeting Monday.

DavidMansolino, 2019-09-25:
We agreed on Monday to:
  1. Update the vehicle robot-window to html.
  2. Drop the bot studio robot-window.
  3. Drop the gui-tracker simulation.
  4. Remove the mechanism that adds lib/webots to the controller lib path (currently used to load Qt).
  5. Keep the darwin-op Qt robot window and add runtime.ini to load Qt.

(please complete the list if I did forget something)

stefaniapedrazzi, 2019-11-25:
@fabienrohrer are you working on this issue?
If yes, could you assign it to you?

fabienrohrer, 2019-11-25:
I'm not working on this, and I'm busy on other projects for now. I tried a bit 1. but did not progress much because I didn't found good ways to merge the vehicle window with the generic one for now. Don't hesitate to take one of these items if you would like to go on with this topic. I think this topic is important.

stefaniapedrazzi, 2019-11-26:
Ok, this week I plan to fix reported issues.
If I have the time I will also address one of these items.

stefaniapedrazzi, 2019-11-27:
I will address point 2. and 3.

stefaniapedrazzi, 2019-11-28:
I will continue with 4 and 5 and dropping the OSM wind and using the runtime.ini for the Fire Bird 6 window.

stefaniapedrazzi, 2019-12-02:
Update:
- we won't drop BotStudio until we have a working alternative
- In Webots R2020a, the vehicle robot window will still use the native Qt robot window and it will then be updated in a future release

So the new tasks for step 1:
- [ ] Remove the documentation of the Qt robot window
- [ ] Remove the mechanism that adds lib/webots to the controller lib path from the generic `Makefile.include`
- [ ] Drop the gui-tracker simulation.
- [ ] Drop the OSM robot window.
- [ ] Update the Qt robot windows Makefile and runtime.ini:
  - [ ] generic robot window
  - [ ] qtutils
  - [ ] vehicle robot window
  - [ ]  darwin-op robot window
  - [ ] Fire Bird 6 robot window.

step 2:
- [ ] Port the vehicle robot window to HTML

stefaniapedrazzi, 2019-12-10:
Step 1 was implemented in #1130.

It remains to port the vehicle robot window to HTML.
I will open a new issue for this.

closed 2019-12-10 by stefaniapedrazzi

DavidMansolino, 2020-01-20:
To me it looks like the two last points from this plan (https://github.com/cyberbotics/webots/issues/893#issuecomment-534954940) are not handled:
  - Remove the mechanism that adds lib/webots to the controller lib path (currently used to load Qt).
  - Keep the darwin-op Qt robot window and add runtime.ini to load Qt.

Did I miss something?

stefaniapedrazzi, 2020-01-20:
The mechanism to link to lib/webots has been removed from the default `Makefile.include`.
But it is still included in the specific Qt robot window Makefiles.

DavidMansolino, 2020-01-20:
Yes the makefile, but these points were referring to the fact that `lib/webots` is in the controller path (in addition to `lib/controller`) which we should at the end remove (to make sure the controller environment is as clean as possible).

stefaniapedrazzi, 2020-01-20:
Ok, I will open a new issue and address these points.

DavidMansolino, 2020-01-20:
Ok, if you prefer, this can also be added to #1170 (as it will be simpler to do this after the removal of the Qt vehicle window).