cyberbotics/webots #912: Links not working in Webots snap package

stefaniapedrazzi, 2019-09-26:
Documentation, keep informed, etc. links are not working.
[image]

stefaniapedrazzi, 2019-09-26:
Same for make animation, make movie and take screenshot: if you answer YES to the message box asking to open the just created file nothing happens.

stefaniapedrazzi, 2019-09-26:
Just a side note in the published snap package R2019b revision 1 the documentation is linking to `master`.

DavidMansolino, 2019-09-26:
I have seen many issues (on Ubuntu 16.04):
  - [ ] issue with animation, movie, screenshot if you want to open it after it has been created (reported by @stefaniapedrazzi too).
  - [ ] Link to online doc doesn't work because tag is not created (reported by @stefaniapedrazzi too).
  - [ ] SUMO doesn't work.
  - [ ] Impossible to recompile the darwin-op robot-windows.
  - [ ] Speaker/sound not working.
  - [ ] E-puck cross-compilation not working.
And some limitations:
  - [ ] matlab not functionnal, we should at least improve the error message.
  - [ ] impossible to use another python version.
  - [ ] impossible to install pip modules.

I have also tested the ROS interface and it is working fine!

stefaniapedrazzi, 2019-09-26:
In my system (Ubuntu 18.04) speaker/sound are working.
About SUMO: `highway_overtake.wbt` simulation is working but not `sumo_interface_example.wbt`.

DavidMansolino, 2019-09-26:
Strange for me I have different issues but both are not working.
`highway_overtake.wbt` the simulation hangs, no new vehicles appear.
`sumo_interface_example.wbt` netconvert fail because of missing libraries.

stefaniapedrazzi, 2019-09-26:
> sumo_interface_example.wbt netconvert fail because of missing libraries.

This is probably the same:
```
[sumo_supervisor] /snap/webots/6/usr/share/webots/projects/default/resources/sumo/bin/netconvert: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
[sumo_supervisor] NETCONVERT failed to generate the network file.
```

DavidMansolino, 2019-09-26:
From Fedora Webots doesn't boot at all:
```
[DavidMansolino@localhost ~]$ webots
Info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Warning: Could not find the Qt platform plugin "wayland-egl" in ""
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

/snap/webots/6/usr/share/webots/webots: line 51:  4091 Aborted                 (core dumped) "$webotsHome/bin/webots-bin" "$@"
```

DavidMansolino, 2019-09-26:
OpenSuse:
   - Installation and launch works fine (excepte that by default the /snap/bin folder is not in the path, but this is a snap issue).
   - [ ] compilation of C/C++ controllers raise exceptions from lsb_release about missing apt-cache
  - Java/Python controller works fine (both runtime and compilation)
  - The open after taking a screenshot/movies/etc. is working fine (which is not working on ubuntu).

I won't test deeper on this platform I just wanted to have an overview of the functionality.

omichel, 2019-09-27:
I just fixed the problems with links (URLs, screenshots, movies, animations) in #855.

omichel, 2019-10-02:
I am closing this issue as the most important bugs are now fixed. I will open a second one with the remaining, less important problems.

closed 2019-10-02 by omichel