cyberbotics/webots #917: Create better splash screen images

omichel, 2019-09-27:
We have 11 different splash screen images. Many of them could be improved, especially, the OP3, Thymio, youBot, PR2, BMW and EOLE to show more PBR effects and a nice environment in the background. Also, new images might be added.

DavidMansolino, 2019-10-04:
Here is what I propose:
  - **bmw**: replace with a tesla in the 'village_center' environment.
  - **EOLE**: in the 'hall' environment (https://cyberbotics.com/doc/guide/samples-environments#hall-wbt).
  - **OP3**: On a soccer field (ideally a nicer one that the current one...).
  - **PR2**: Same screenshot but in the kitchen environment (https://cyberbotics.com/doc/guide/samples-environments#kitchen-wbt)
  - **youBot**: in the pick and place world environement.
  - **Thymio**: in the 'complete_appartement** environment (https://cyberbotics.com/doc/guide/samples-environments?version=develop#complete_apartment-wbt)
  - **New**: 
    - abb: https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_abb/abb.png

DavidMansolino, 2019-12-12:
Fixed in  #1163

PR #1163 (2019-12-11): Improve Splash Screen
Fix #917
closed 2019-12-12 by DavidMansolino