cyberbotics/webots #927: Extend wb_supervisor_simulation_reset();

DavidMansolino, 2019-09-30:
Many users are asking for a way to rest the simulation but not the supervisor controller.
A simple solution would be to add an argument (possibly optional) to the wb_supervisor_simulation_reset function to say if the supervisor itself should be restarted or not.

simonbogh, 2019-11-24:
I vote for this, and additionally, if pausing the simulation with wb_supervisor_simulation_set_mode, make it possible to take one simulation step. 

This would be beneficial in cases where you are doing numerical optimization through e.g. ROS and you need the state information at each timestep and you want to control when each timestep happens. 

I do not know how much it takes, if possible, to get this kind of functionality, but it would be very useful for deep learning purposes.

fabienrohrer, 2019-11-25:
Just some considerations about `wb_supervisor_simulation_reset()` to do this in practice:

- another solution would be to add a flag to prevent the reset all the controllers (not only the supervisor). But this seems to be a less good idea, because it's harder to find use cases. Generally a single supervisor leads the numerical optimization, the other controllers are slaves.
- the flag could be optional. pros: more concise API / backward compatible (the users using this are power users who can deal with this issue easily). cons: the C API would be ugly, users using this in a supervisor should determine if they want to reset the current supervisor or not.

Therefore, I would go for this function: 

```
wb_supervisor_simulation_reset(bool reset_self);
```

omichel, 2019-11-29:
While breaking the C API, I believe we should consider also the case where we don't want to restart the other controllers as I believe there are common use cases for that in evolutionary computation for example when a supervisor is continuously communicating with robot controllers. So, I would rather propose this function:

```
wb_supervisor_simulation_reset(bool restart_self, bool restart_others);
```
With both parameters defaulting to TRUE in the C++/Python/Java APIs, so that don't break the backwards compatibility.

DavidMansolino, 2019-11-29:
In that case, why not doing something more generic, like setting the argument to be an array of WbNodeRef (which should correspond to robot nodes for which we don't want to restart the controller)?

```wb_supervisor_simulation_reset(WbNodeRef *keep_controller_running, int keep_controller_running_size);```
(argument name are just to explain, we should find better ones)

omichel, 2019-11-29:
Or even simpler, change the behavior of `wb_supervisor_simulation_reset()` to not restart any controller at all and use the `wb_supervisor_node_restart_controller()` function right away to restart only the controllers we want to restart. That makes the API simpler, powerful and cleaner IMO.

DavidMansolino, 2019-11-29:
Indeed, that's very simple and powerful, the only drawback I see is that it will not correspond anymore to the reset from the GUI (but if it is correctly documented that's probably fine).

omichel, 2019-11-29:
Yes, we should document clearly that the reset from the GUI is a combination of `wb_supervisor_simulation_reset()` and `wb_supervisor_node_restart_controller()` on every robot.

fabienrohrer, 2019-11-30:
That's sounds a much better solution to me too.

DavidMansolino, 2019-12-10:
Fixed in #1145.

closed 2019-12-10 by DavidMansolino