cyberbotics/webots #928: The buttons of the web interface look bad

omichel, 2019-10-01:
The height of the fast button is not the same as the other run and step buttons.
The shape of the run button is different from the shape of the other buttons, etc.
[image]
We should reuse the icons of Webots which are better designed. That would provide also more consistency.
[image]

stefaniapedrazzi, 2019-10-01:
When the fast button has been added it seems that has been copied directly from the Webots icons without taking into account the style used in the web interface.

stefaniapedrazzi, 2019-10-01:
I also don't like very much the appearance of the new world selection drop-down list and I think it should be improved.

closed 2019-10-15 by stefaniapedrazzi