cyberbotics/webots #939: Lens flare is buggy

omichel, 2019-10-01:
As it can be seen in `projects/vehicles/worlds/highway.wbt`, the lens flare seems wrong:

[image]

fabienrohrer, 2019-12-03:
I can indeed reproduce this issue:

[image]

Some times, but rarely, it's working fine after reverting the world, or loading another world before:

[image]

This means it's caused by bad LenseFlare parameters (as I originally thought) but with a real bug. I have to check if this post-processing is applied in the correct target.

omichel, 2020-12-08:
A nice implementation of lens flare in a shader that works with WebGL is provided here: https://www.shadertoy.com/view/4sX3Rs
We should try to implement it in Webots and see whether that resolves this bug.

omichel, 2020-12-09:
Removing the `noperspective` keyword on the R2020b-rev2 `lens_flare.*` shaders didn't break the lens flare.
It seems they were already broken on the develop branch for another reason.

omichel, 2020-12-09:
I was wrong, the lens flare are still working on the current develop branch. They are just a little bit tricky to observe (we have to move the viewpoint, turn on the lights of the car to see them well):
[image]