cyberbotics/webots #943: Cannot open a "Support ticket" on my MacBook Air

fabienrohrer, 2019-10-01:
On my MacBook Air only, the "Help / Support Ticket (Premier Service)" item fails with this error dialog:

[image]

.. while the other URLs are working.

fabienrohrer, 2019-10-01:
The URL is:

`https://cyberbotics.com/support_ticket.php?&os=Darwin 18.7.0 x86_64&graphics=Intel Inc. - Intel HD Graphics 4000 OpenGL Engine - 4.1 INTEL-12.10.12&version=R2019b revision 1&type=ticket`

.. and works when typed directly in a browser.

omichel, 2019-10-01:
Maybe we should URLencode the string (e.g., the spaces should be `%20` instead of ` `).

omichel, 2019-10-01:
I tested on Windows and it works.

omichel, 2019-10-01:
Maybe you should try to remove the `StrictMode` from here: https://github.com/cyberbotics/webots/blob/revision/src/webots/core/WbDesktopServices.cpp#L31

fabienrohrer, 2019-10-01:
Good point, it seems to solve my issue.
(I can reproduce the issue in my development environment by copying the string above.)

It seems that this URL contains something which is not liked by the strict mode.

omichel, 2019-10-01:
Yes, but the problem is specific to macOS...

fabienrohrer, 2019-10-01:
Ok, the issue appears when the strict mode is used and the url contains spaces instead of `%20`.

fabienrohrer, 2019-10-01:
As this mechanism is used in different contexts, for example to open links from the documentation or from the robot window, I expect that some tolerance should be acceptable (for example is users change this). I think that removing the "strict" mode everywhere will be more robust.

closed 2019-10-01 by fabienrohrer