cyberbotics/webots #975: Webot 2019b Rv1 crash when starts at windows 10 pro

fgw, 2019-10-05:
**Describe the Bug**
Webot 2019b Rv1 crash when starts at windows 10 pro. It crashs when finizaing the note at 99%.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
windows 10 pro
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]
Nvidia GeForce 940M ( 2 GB

**Additional context**
Add any other context about the problem here.

电脑基本信息

我的电脑	联想 80R5 笔记本电脑

操作系统	Windows 10 64位

主显卡	独立显卡(对游戏和电影支持较好)

IE浏览器	版本号  11.0

基本硬件展示

处理器	英特尔 Core i7-6500U @ 2.50GHz 双核

主板	联想 Yoga 500-14ISK

内存	8 GB ( 海力士 DDR3L 1600MHz )

主硬盘	西数 WDC WD10SPCX-24HWST1 ( 1 TB / 5400 转/分 )

主显卡	Nvidia GeForce 940M ( 2 GB / 联想 )

显示器	奇美 CMN14B1 ( 14 英寸  )

网卡	英特尔 Dual Band Wireless-AC 3165

声卡	瑞昱  @ 英特尔 High Definition Audio 控制器

DavidMansolino, 2019-10-07:
@fgw do you experience the same crash with Webots R2019b ?
Have you tried the safe mode (https://cyberbotics.com/doc/guide/starting-webots#safe-mode) ?
Are you experiencing this on all the worlds or on specifics ones?

DavidMansolino, 2019-10-30:
Without any other information we will not be able to help you. Please do not hesitate to re-open the issue if you have more information to provide.

fgw, 2019-11-14:
Hi David,

Sorry to reply you late. 
I figured out the cause of this issue.  When I used the Intel integrate graphics card, the Webot can start normally. But when I configured the PC to use the Nvidia graphics card, the Webot has the same issue.

My system is windows 10 pro 64 bit 10.0.18362.  Intel graphics card driver version 21.29.16.4821. Nvidia graphic driver version 441.20, graphic card is Geforce 940M.

I duplicated the issue.  Please refer the video as below:

Error happened with Nvidia graphics card:
https://drive.google.com/file/d/1377YJ_-86TF3Ymf97vau1y5JUWcVEYdC/view?usp=sharing

Normal with Intel graphics card:
https://drive.google.com/file/d/1WRDAatj1NBaKaIrpk6TX78sFEB2yAzB3/view?usp=sharing

Regards,
Guangwei

DavidMansolino, 2019-11-15:
Hi @fgw, thank you for the update.

Have you tried the safe mode (https://cyberbotics.com/doc/guide/starting-webots#safe-mode) with the Nvidia GPU?

fgw, 2019-11-18:
> Hi @fgw, thank you for the update.
> 
> Have you tried the safe mode (https://cyberbotics.com/doc/guide/starting-webots#safe-mode) with the Nvidia GPU?

I tried Webots safe-mode. The Webots can start normally with Nvidia GPU.

fabienrohrer, 2019-11-18:
Good news!

You should be able now re-start Webots regularly (i.e. without the safe-mode) using you NVIDIA GPU.

It would be awesome if could help us to determine which OpenGL preference is causing the crash. To do so just increase the OpenGL settings in the Webots preferences, restart Webots after each modification, until it crashes again. I typically suspect the "ambient occlusion" or the "texture quality preference". Use the safe mode to restore the preferences at their lowest level.

fgw, 2019-11-25:
> Good news!
>
> You should be able now re-start Webots regularly (i.e. without the safe-mode) using you NVIDIA GPU.
>
> It would be awesome if could help us to determine which OpenGL preference is causing [...]

Sorry, I am quite busy these days. I will try the configuration later.

mohakhalili71, 2019-11-27:
Hi guys

I had the same problem, and read your comments.

Actually, I experience the same problem like @fgw , i.e. when I used the Intel integrate graphics card, the Webots can start normally. However, the world was not displayed properly and the graphics were poor. But when I configured the PC to use the Nvidia graphics card, the Webots has the same issue.

My configuration:
System is a Thinkpad e440 
windows is 10 pro
Intel graphics card is Intel(R) HD Graphics 4600
Nvidia graphics card is Geforce 840M

According to @fabienrohrer, I started Webots by safe-mode. The Webots can start normally with Nvidia graphics card and after that, I re-start Webots regularly (i.e. without the safe-mode) using my NVIDIA GPU and Webots opened without crash.

Then I tried to understand which OpenGL preference is causing the crash. To do so according to @fabienrohrer again :), first I held the texture quality preference as low and started to increase ambient occlusion preference step by step and then re-started the Webots, in this case there was no crash in all of the steps from low to ultra changes in ambient occlusion preference.
 
Later I held ambient occlusion preference as disabled and started to change texture quality preference, I changed texture quality preference as medium and re-started the Webots, there was no crash. After that, I changed texture quality preference as high and re-started the Webots, WOW it caused that crash happened.

I can restore the preferences at their lowest level by safe mode again and apply the changes in combination (for example ambient occlusion = low, change the texture quality or vice versa) and let you know the result, if you need, so let me know.

Finnaly, I hope that I have helped to improve the Webots :)

DavidMansolino, 2019-11-27:
Hi @mohakhalili71 thank you for your tests, it indeed helps a lot.
We have improved in some way the textures management in our development version, may I ask you to try with a beta of R2020a if it works fine for you with all the OpenGL features at maximum?
You can download it from here: https://github.com/cyberbotics/webots/releases/tag/nightly_26_11_2019

fabienrohrer, 2019-11-27:
@mohakhalili71 Thank you for these precise tests, it's really helpful. I agree with @DavidMansolino, this is certainly a GPU memory overload (we should improve our tests when instantiating the textures accordingly, if possible). Currently, the `develop` version consumes indeed less memory, and it's likely that the crash won't occur (but it is only pushed further).

mohakhalili71, 2019-11-27:
Hi @DavidMansolino 

Why not!
So give me time to test it, I'm in Iran.

mohakhalili71, 2019-11-27:
@stefaniapedrazzi your welcome :)
Hopefully the crash won't happen in R2020a :)

mohakhalili71, 2019-11-27:
Hey guys,

I installed R2020a and it ran without any crashes. But now I have another problem that mentioned below in picture. When I open my project and run it, this error appears,

[image]

and in console I see :

[e-puck_go_forward] Traceback (most recent call last):
[e-puck_go_forward]   File "e-puck_go_forward.py", line 5, in <module>
[e-puck_go_forward]     from controller import Robot, Motor
[e-puck_go_forward]   File "C:\Users\Moha\AppData\Local\Programs\Webots\lib\controller\python37\controller.py", line 15, in <module>
[e-puck_go_forward]     import _controller
[e-puck_go_forward] ImportError: DLL load failed: The specified module could not be found.
WARNING: 'e-puck_go_forward' controller exited with status: 1.

What should i do now?
Is your suggestion to use the same stable version?
Or is there a solution for this problem?

DavidMansolino, 2019-11-27:
Thank you for the confirmation, at least it does confirm that the crash due to GPU is gone.

mohakhalili71, 2019-11-27:
@DavidMansolino @fabienrohrer Thank you for your help.

DavidMansolino, 2019-11-27:
@mohakhalili71 you're welcome.
About the issue you are experiencing with R2020a, I have found the cause of the problem and it is going to be fixed in #1131

DavidMansolino, 2019-11-27:
@mohakhalili71 you will be able to download a new beta of R2020a containing the fix from tomorrow morning on this page: https://github.com/cyberbotics/webots/releases

DavidMansolino, 2019-11-28:
The beta of R2020a with both bugs fixed is now available.
I allow myself to close this issue as both bugs seems fixed. But feel free to post new comments if there is some news.

closed 2019-11-28 by DavidMansolino

MyKoo, 2020-01-08:
Hi everyone!
I have crash on start up with 2020a version. Launching it with safemode on does nothing. Do you have any ideas how I can repair it?