cyberbotics/webots #986: Madatory npm Dependency

DavidMansolino, 2019-10-08:
**Describe the Bug**
The npm dependency is documented as optional: https://github.com/cyberbotics/webots/wiki/Linux-Optional-Dependencies#nodejs

However, the main compilation process fails if npm is not installed. It should just be skipped if npm is missing, as it is already the case for java and python for example.

stefaniapedrazzi, 2019-10-24:
I just checked on Ubuntu 18.04 and it works correctly for me on revision even when building from the main `webots` folder: if `npm` is not found, the build of `webots.min.js` is skipped.

DavidMansolino, 2019-10-24:
I just retried both on a clean VM and in a Docker (where I originally found the issue), but I can't reproduce this anymore, probably a wrong setup on my first trial.

PR #1042 (2019-10-28): Fix cleanse rule if npm is not installed
Fix #986 😸
closed 2019-10-24 by DavidMansolino