cyberbotics/webots #993: Improve environment variables instruction for extern controllers

stefaniapedrazzi, 2019-10-10:
In the environment variables section of the extern controllers:
https://www.cyberbotics.com/doc/guide/running-extern-robot-controllers#environment-variables
there are many mixed information about different languages and platforms and it is not so straightforward to understand what should be set.

Then there is a generic table followed by specific instructions for C/C++ and Java, but not for Python.
I understand that Java needs some additional instructions but not why there are instructions for C/C++ (same as what is already given in the table) if for the other languages we should only use the table.

Maybe it would be better to create some tabs (like for the nodes API) to choose the platform and/or the language so that there will be less but more relevant instructions.

omichel, 2019-10-10:
Unfortunately, we cannot create tabs for both the platform and the language. So, we should choose or make sections...

stefaniapedrazzi, 2019-10-10:
I would make tabs for platforms.

stefaniapedrazzi, 2019-10-10:
I commit a first version on the branch `fix-extern-controller-on-snap` (still have to fix an issue with the new line just after the first new table)
https://www.cyberbotics.com/doc/guide/running-extern-robot-controllers?version=fix-extern-controller-on-snap#environment-variables

omichel, 2019-10-10:
Unfortunately, it doesn't seem our doc tab system is well suited for handling other things than languages: when selecting a combination of OS and languages and reloading the page, we loose part of the selection...

stefaniapedrazzi, 2019-10-10:
oh yes, I didn't try to reload the page.
I will change back the second table to a list.

stefaniapedrazzi, 2019-10-10:
I just updated the page

stefaniapedrazzi, 2019-10-10:
Ok, so I'm running out of proposal.
I will revert the changes.

The platform table is very handy and cleans up a lot useless informtation on the page.
The language table was better too. But from my point of view less important than the platform one.

stefaniapedrazzi, 2019-10-10:
On the mid term we should improve the doc mechanism to handle the tabs.

stefaniapedrazzi, 2019-10-24:
After merging #1025 it will be possible to use multiple tables.

stefaniapedrazzi, 2019-10-24:
Done in #1025.

closed 2019-10-24 by stefaniapedrazzi