cyberbotics/webots #998: Wrong redirection of Google searches

michou214, 2019-10-14:
**Documentation page(s) concerned**
https://www.cyberbotics.com/doc/reference/proto-scoping-rules?version=R2018a

**What is the problem with the current documentation?**
By searching the keyword "_webots proto scoping rules_" on Google, an empty documentation page is proposed. The problem is that the proposed link makes a reference to an earlier version of the Webots documentation. In the mentioned url, we can see that there is a version **?version=R2018a**. If this part is deleted, the right page with the content of the documentation is displayed.

**Describe how you would fix it**
As all the other pages of the PROTO documentation are correctly referenced, I think there is a redirection error to fix.

omichel, 2019-10-14:
I asked google to index the correct page instead. Let's check this back in a few days to see if google indexed the correct page.

michou214, 2019-10-21:
I just try now and the problem is fixed.

closed 2019-10-21 by DavidMansolino