cyberbotics/webots #1016: Cannot connect to Webots on pipe

UPUPGOO, 2019-10-17:
I want to control my robot by using Pycharm. I followed the [documentation](https://cyberbotics.com/doc/guide/using-pycharm-with-webots), but I got some errors.

Cannot open pipe file: \\.\pipe\webots_8176_877243746
Cannot connect to Webots on pipe: \\.\pipe\webots_8176_877243746

So, I would appreciate any help.

DavidMansolino, 2019-10-17:
Is Webots already started when you launch the extern controller?
Are you using the latest version of Webots (R2019b-rev1)?

UPUPGOO, 2019-10-24:
Sorry for the late reply. I am using the latest version of Webots (R2019b-rev1) and started the Webots when I launch the extern controller. The error is now gone, although I didn't know the reason. By the way, when there are a lot of background programs, Webots will inexplicably die.

DavidMansolino, 2019-10-24:
OK that's strange, just for curiosity before getting the `Cannot open pipe` error, were you getting some `Cannot open file: ... (retrying in X seconds)` warnings?

About your strange Webots die when you have a lot of background programs, are you able to identify one specific program that is causing this? What is your OS? How much RAM do you have? What is the output given by Webots when you start it with the `--sysinfo` command?

UPUPGOO, 2019-10-25:
No 'Cannot open file: ... (retrying in X seconds)' warnings.
When I reduce the number of background programs, Webots works well. I tried to use Webots on my laptop(HP ZHAN66 Pro G1) and it inexplicably died too, but Webots woked well on Ubuntu16. The configuration of my laptop is Win10, 8GB RAM and MX150. The configuration of my desktop computer is Win10, 8GB RAM and GT960.

DavidMansolino, 2019-10-30:
You should probably check if you run out of memory or GPU memory when you have several background programs.

UPUPGOO, 2019-11-01:
> You should probably check if you run out of memory or GPU memory when you have several background programs.

Thank you. Whatever Webots is really helpful.

closed 2019-11-01 by UPUPGOO