cyberbotics/webots #1054: Numbered item list looks bad in the online documentation

omichel, 2019-11-01:
This screenshot was taken from Firefox / Windows 10 at https://cyberbotics.com/doc/guide/using-visual-studio-with-webots
[image]
The number should be on the same line as the text like in markdown:
1. first item.
2. second item.

DavidMansolino, 2019-11-04:
Same issue on Firefox / Ubuntu 16.04

fabienrohrer, 2019-11-06:
This is because of main.css which bolds counter items without class encapsulation:

```
ol > li {
  list-style-type: none;
  counter-increment: item;
  text-indent: -1em;
}
ol > li:before {
  font-weight:bold;
  padding-right: 0.5em;
  content: counter(item) " ";
}
```

fabienrohrer, 2019-11-06:
1. A simple solution would be to drop these 2 rules (and so to drop the bold on the counters).
2. A better solution would be to introduce classes in index.php, at least for the lists. This is not in my hands...
3. A poor solution would be to override these rules in `webots-doc.css`.

omichel, 2019-11-06:
I don't understand what you mean by classes in `index.php`. Are you referring to `doc.php`? What kind of class do you mean?

fabienrohrer, 2019-11-06:
No, I mean the main CSS of the cyberbotics website. For example here, it replaces all the `ol > li[:before]`. This causes conflicts in the doc. The doc is build with HTML classes, and the main web site not. Adding a classes in cyberbotics.com/index.php and main.css is the best solution.

fabienrohrer, 2019-11-06:
We just discussed with Olivier, and we will go for solution 1.

fabienrohrer, 2019-11-06:
Fixed by overriding main.css without the 2 problematic rules.

closed 2019-11-06 by fabienrohrer