cyberbotics/webots #1055: Improve HTML robot-window Doc

DavidMansolino, 2019-11-01:
**Documentation page(s) concerned**
https://www.cyberbotics.com/doc/guide/controller-plugin#robot-window

**What is the problem with the current documentation?**
The description on this page is way not sufficient, we should write a small how-to or improve this doc to help users creating their own robot-window.
Furthermore, instead of pointing to the thymio robot window we should point to the 'custom_robot_window' sample one.

chaytanyasinha, 2020-05-18:
@DavidMansolino May I take up this issue?

DavidMansolino, 2020-05-18:
Yes, of course, you are welcome to contribute!

chaytanyasinha, 2020-05-21:
@DavidMansolino @omichel Can you please tell me are there any resources to refer while writing the how to section there

chaytanyasinha, 2020-05-23:
@DavidMansolino @omichel Can you please tell me how should I point to the 'custom_robot_window' sample one?

chaytanyasinha, 2020-05-29:
@DavidMansolino Should I just mention about the C file and JavaScript file for the how-to section?
Will that be all?

DavidMansolino, 2020-05-29:
Mentioning is not sufficient, you should explain how they work and how to create them.

chaytanyasinha, 2020-05-29:
@DavidMansolino Yeah I meant that only. I meant that steps from c file and JavaScript file only I have to explain.Not any thing about html and css?

DavidMansolino, 2020-05-29:
It worse mentioning them too, but they are less important indeed.

chaytanyasinha, 2020-05-29:
@DavidMansolino I got it.Thanks

stefaniapedrazzi, 2020-08-19:
At this point it is probably better to write directly an example up-to-date to the new robot window file structure added in #2096.

lukicdarkoo, 2021-01-11:
@stefaniapedrazzi 
What I noticed from the questions on Discord is that:
- We should have a very simple example that is self-explanatory.
- The example should be placed in a dedicated folder.
- The example should be implemented in Python.

The closest example we have so far is `projects/samples/howto/plugins/robot_windows/custom_html_robot_window`, but there the following problems:
- It doesn't work! We should fix it. We are missing script tag, HTML tag (#2096)...
- It is in C, not Python.
- It is not simple. I would rather just make a counter to demonstrate how the robot window and the controller can communicate. And point to the Thymio example as the next step.
- It is not placed in a dedicated folder, meaning that the users cannot easily understand the structure.
- I am not sure whether it is a good idea to promote having `custom_html_robot_window.c`. Then, there are two controllers (typical and robot window controller) for the single robot. I believe it is better to promote the idea that the typical controller should communicate with the robot window. It is simpler and it cannot lead to undesired behavior.

stefaniapedrazzi, 2021-01-11:
I agree with your comments and in fact this issue is exactly about improving the overall documentation about the robot window.

* of course if the already available example example doesn't work, then it has to be fixed because it is a bug or replaced or removed. We didn't carefully tested when releasing the last version.
* the problem between Python/C is easy to solve and we could provide both controllers quite easily.
* we can also refer to the Thymio but the Thymio one is quite complex and I doubt that can really help users.
* as described in this issue description the best would probably be to write a new tutorial explaining how to build step by step a robot window. Then we could put the related source code in the tutorials folder and create a dedicated folder there.
* I think it is important to also describe how to create a C program unrelated to the controller for advanced users.
This is a quite important functionality that will simplify reusing the same robot window independently from the controller.
Everything that is not documented would never be used.
But we can make two different simple examples in the tutorial one sending information to the robot window from the controller and one directly from the robot window C program.

The already available custom HTML is very simple. The only "complex" part is the C program.
I would not trash and create a new (or multiple) one but rather fix it and create two different versions:
* one controlled directly from the controller program (in C/Python)
* one controlled from the robot window C program
Then I would also write a tutorial explaining how they are built.

stefaniapedrazzi, 2021-01-11:
> I am not sure whether it is a good idea to promote having custom_html_robot_window.c. Then, there are two controllers (typical and robot window controller) for the single robot. I believe it is [...]

Communicating directly from the controller has the big limitation that you have to readd all the robot window communication code to any controller you may use. This is the reason why we also added the possibility to write directly a robot window C program.

lukicdarkoo, 2021-01-11:
> The already available custom HTML is very simple. The only "complex" part is the C program.
I would not trash and create a new (or multiple) one but rather fix it and create two different versions:

I would say there is a lot of noise. I believe the initial example should demonstrate the two-way communication between the robot window and the controller as simple as possible. It should be enough for users to understand how to build upon that. I believe it will cover most use-cases.

Then, we can extend the `custom_html_robot_window` example as you suggested:
> - one controlled directly from the controller program (in C/Python)
> - one controlled from the robot window C program

stefaniapedrazzi, 2021-01-11:
> I would say there is a lot of noise.

Honestly I don't see where is the noise.
There is just a console, a sensor display and a motor control function and these are the three common functionalities a robot window should have.

However, the important thing is to add a clear documentation about how to create robot windows. So, if you want to address this issue, please go ahead in the way you think it is the best one. 
We should just really pay attention to not have multiple duplicated examples to maintain.

lukicdarkoo, 2021-01-11:
@stefaniapedrazzi I was thinking something like https://github.com/cyberbotics/webots/pull/2639/commits/307dd3a0b5bbcd4e41ab690d72334a8ac33cfaaf is much simpler than the `custom_html_robot_window` example.

lukicdarkoo, 2021-01-11:
> We should just really pay attention to not have multiple duplicated examples to maintain.

> I think it is important to also describe how to create a C program unrelated to the controller for advanced users.
This is a quite important functionality that will simplify reusing the same robot window independently from the controller.
Everything that is not documented would never be used.
But we can make two different simple examples in the tutorial one sending information to the robot window from the controller and one directly from the robot window C program.

I completelly agree with these points, but I would also like that the users have a "Hello World" example, something like 307dd3a

omichel, 2021-01-11:
I believe the simple example developed in #2639 in a good introduction to robot windows. But I would also keep a more elaborated example (including also a C controller). So, I think we can afford to maintain two examples for robot windows.

PR #2639 (2021-01-13): Improve robot window documentation
Fixes #1055

**Documentation**
- https://cyberbotics.com/doc/guide/samples-howto?version=feature-robot-window-doc#custom_robot_window_simple-wbt
- https://cyberbotics.com/doc/guide/controller-plugin?version=feature-robot-window-doc#robot-window

**TODO**
- [x] Add a very basic robot window
- [x] Refactor the `projects/samples/howto` directory, so each example is in a dedicated folder
  - [x] Change the documentation accordingly
- [x] Delete extra Makefiles (e.g. `projects/samples/howto/plugins/physics/Makefile`) to minimize the noise
- [x] Fix the `custom_htm [... cut]
closed 2021-01-13 by lukicdarkoo