cyberbotics/webots #1057: Web export problem with Firefox under Linux

Pouya-moh, 2019-11-01:
Hi. I am rather new to webots so my apology if the issue is a non-issue.

I downloaded the latest 2019b-rev1, extracted it, opened a demo scene (stewart platform), recorded a HTML5 video and stopped. Webots automatically triggered my Firefox and opened the scene, however, there is big black box there. No scene, no player, nothing.

I tested my browser's support for x3d in several other places and they all work. I opened the exported files in a Firefox under virtual Windows 10 and it worked there as well. 

Firefox web console gives some hints (not sure if relevant)
```
window.controllers/Controllers is deprecated. Do not use it for UA detection
THREE.WebGLRenderer 104
X3D: Loading stewart_platform.x3d
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at file:///to-my-exported-files

```
**System**
 - Ubuntu 18.04
 - Nvidia GeForce GTX 1050 Ti (GPU 0), Driver ver. 390.116
 - Firefox 70.0

Thanks!
P.

stefaniapedrazzi, 2019-11-04:
Yes, the problem is the Cross-Origin Request Security (CORS).

Thank you for reporting this issue, this is new in Firefox 68 and later and we didn't notice it yet.
https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS/Errors/CORSRequestNotHttp
I will check if it is possible to fix it directly in the generated web model or update the documentation.

**Workaround:**

Here is explained how to disable completely the CORS `security.fileuri.strict_origin_policy` in Firefox:
http://testingfreak.com/how-to-fix-cross-origin-request-security-cors-error-in-firefox-chrome-and-ie/

But to let the web model/animation/streaming working is also enough to disable the `privacy.file_unique_origin` option.

stefaniapedrazzi, 2019-11-05:
I improved the documentation and added the instructions to solve this error.
Please re-open this issue if you still have any problem.

closed 2019-11-05 by stefaniapedrazzi