cyberbotics/webots #1064: OpenGL error: GL_INVALID_OPERATION

ibrahimqazi, 2019-11-05:
When I run Webots from terminal, I get a lot of messages:

> OpenGL error: GL_INVALID_OPERATION  
What is the possible solution to this error ?

DavidMansolino, 2019-11-05:
Does this happens with one specific world or with all of them?
What is your GPU and are you GPU drivers up to date?
And what is your OS?

ibrahimqazi, 2019-11-05:
@DavidMansolino I am using a single world with different names. My GPU is NVIDIA 950M, my drivers are up to date, Ubuntu 16.04

DavidMansolino, 2019-11-06:
Ok perfect, would you agree sharing this world file with us so that we can try?

ibrahimqazi, 2019-11-11:
#VRML_SIM R2019b utf8
WorldInfo {
  basicTimeStep 16
}
Viewpoint {
  orientation -0.34658035748168825 0.8688419281702092 0.3535473937979718 1.7132539825092965
  position 5.09275205892429 5.802982367713556 0.6552785994204011
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Floor {
  size 3 3
}
WoodenBox {
  translation 0.0888639 0.06 0.285177
  rotation 0 1 0 1.54462
  size 0.1 0.1 0.1
}
WoodenBox {
  translation 0.0927326 0.05 -0.271753
  rotation 0 1 0 3.11541
  name "wooden box(1)"
  size 0.1 0.1 0.1
}
E-puck {
  controller "e-puck_control_speed"
}
Solid {
  translation 0 0.2 -0.2
  rotation 0 1 0 -0.261795307179586
  children [
    DEF BALL_GEOMETRY Shape {
      appearance PBRAppearance {
        roughness 1
        metalness 0
      }
      geometry Sphere {
        radius 0.06
        subdivision 2
      }
    }
  ]
  boundingObject USE BALL_GEOMETRY
  physics Physics {
  }
}
DEF WALL_SHAPE Wall {
  translation 0 0 1.45
  size 3 0.3 0.1
  appearance Roughcast {
    colorOverride 0.4 0.5 0.5
    textureTransform TextureTransform {
      scale 1 2.4
    }
  }
}
DEF WALL_SHAPE Wall {
  translation 0 0 -1.45
  name "wall(1)"
  size 3 0.3 0.1
  appearance Roughcast {
    colorOverride 0.4 0.5 0.5
    textureTransform TextureTransform {
      scale 1 2.4
    }
  }
}
DEF WALL_SHAPE Wall {
  translation -1.46 0 0.0400031
  rotation 0 1 0 1.5708
  name "wall(2)"
  size 2.9 0.3 0.1
  appearance Roughcast {
    colorOverride 0.4 0.5 0.5
    textureTransform TextureTransform {
      scale 1 2.4
    }
  }
}
DEF WALL_SHAPE Wall {
  translation 1.46 0 5.19723e-15
  rotation 0 1 0 -1.5707953071795862
  name "wall(3)"
  size 2.9 0.3 0.1
  appearance Roughcast {
    colorOverride 0.4 0.5 0.5
    textureTransform TextureTransform {
      scale 1 2.4
    }
  }
}

ibrahimqazi, 2019-11-11:
one of my world file. @DavidMansolino please have a look

stefaniapedrazzi, 2019-11-18:
I ran you simulation on a Ubuntu 16.04 machine with a NVIDIA GeForce GT 640M graphics card and I don't get any OpenGL errors.

Note that you did not provide the controller `e-puck_control_speed`.
Did you get the OpenGL errors even setting the e-puck controller to `void`?

Could you please post the output of the command `webots --sysinfo`?

omichel, 2019-11-29:
I assume the problem is fixed in the pre-release version of Webots R2020a, right? I am therefore closing this issue. Feel free to re-open it if that's not the case.

klues, 2020-03-31:
I'm facing a similar issue, also on an Ubuntu 16.04 virtual machine, Webots `R2020a revision 1`.
I'm opening the file [obstacle_avoidance.wbt](https://github.com/cyberbotics/webots/blob/master/projects/samples/robotbenchmark/obstacle_avoidance/worlds/obstacle_avoidance.wbt) and it looks like this: http://prntscr.com/rq2qnx -> many objects in the 3D view are missing and `OpenGL error: GL_INVALID_OPERATION` errors on console.

The actual `obstacle_avoidance.wbt` file I'm opening is [this](https://gist.github.com/klues/9e60b2bd7465f159968fefe8b964db7f) (in case that the current file in `master` is different).

Interesting behaviour: if the world is loaded on starting Webots, I get this error, if I do `File -> Reload World` afterwards, everthing is looking fine: http://prntscr.com/rq2wdt

Any ideas?

DavidMansolino, 2020-03-31:
Do you have the contact point optional rendering enabled?
If yes, this is a known bug that we are currently trying to fix: #1434

klues, 2020-03-31:
If you are refering to the option in `View -> Optional Rendering`: no. No option is enabled there.

DavidMansolino, 2020-03-31:
Ok, in that case it is probably due to the Virtual Machine, I can't reproduce the issue on a native Ubuntu 16.04.

klues, 2020-03-31:
Ok, thanks for the information. I can't imagine why it then would work on `Reload world`, if the virtual machine would be the problem.

klues, 2020-03-31:
Other worlds, e.g. `tower_of_hanoi.wbt` are working everytime, also on startup.

DavidMansolino, 2020-03-31:
I just tried the ``obstacle_avoidance.wbt`` world on a Virtual Machine of Ubuntu 16.04 and unfortunately not openGL error for me.
May I ask you to try launching Webots with the ``MESA_DEBUG`` environmnent variable set (`export MESA_DEBUG=1`) to try to get more meaningful error messages?

klues, 2020-04-01:
Thanks for investigating! What kind of VM software did you use? I'm currently using VMware Player 14.

Here is the log with `MESA_DEBUG` enabled: [log.txt](https://gist.github.com/klues/d07b4e604cb95c48505ac81faf513a9d)
Basically these are the errors:
```
Mesa: User error: GL_INVALID_ENUM in glGetIntegerv(pname=GL_TEXTURE_FREE_MEMORY_ATI)
function is no-op
Mesa: User error: GL_INVALID_OPERATION in glDrawElements(no VAO bound)
OpenGL error: GL_INVALID_OPERATION
```

DavidMansolino, 2020-04-01:
> What kind of VM software did you use

That can explain the difference, I am using VirtualBox.
Thank you for the errors, we might be able to find where it does come from with this. In the meantime, you can safely ignore them.

klues, 2020-04-01:
I just tried Ubuntu 16.04 on latest VirtualBox and there webots didn't start at all because of "no OpenGL 3.3" support.

Maybe my hardware is just not good enough, I don't have a dedicated graphics card, but only a standard Intel onboard graphics chip.

DavidMansolino, 2020-04-01:
That may be a reason indeed, you should probably try changing the graphics controller in the settings of your virtual machine.

DavidMansolino, 2020-04-01:
In any case it is recommended to avoid using a virtual machine but install Webots on your native OS.

klues, 2020-04-01:
In the settings of the virtual machine I've already set up the maximum resources for graphics.

Yeah, I know that using the native OS would be much better, but I'm preparing an online teaching course for ROS and I assume that most of my students won't have a native Linux installed and I think ROS + Webots on Windows also won't work very well. So the idea is to provide a Linux VM to them.

It's a pity since Webots in the VM generally works quite well, despite of this error. However I have to assume that many of my students will have even less capable hardware than my notebook and therefore I don't know if Webots in the VM is any good idea.

Do you know any other lightweight robot simulator with very simple graphics or maybe even 2D graphics and that therefore will run smoothly on any VM/OS?

DavidMansolino, 2020-04-01:
> In the settings of the virtual machine I've already set up the maximum resources for graphics.

It is not a matter of setting to maximum, it is a question of selecting the graphics controller that emulates the GPU, they are several options, some of them are working better than others, for example I am using `VMSVGA` which works fine for me.
In any case this error message should not prevent you from using Webots, it is just a message you can safely ignore it.
To get better performance in a VM I would also recommend using the safe mode of Webots: https://www.cyberbotics.com/doc/guide/starting-webots#safe-mode

klues, 2020-04-01:
Yeah, in VirtualBox I also tried `VMSVGA`, but there I got the `OpenGL 3.3 is not supported` error on starting Webots. In VMware Player there isn't an option like that.

Thanks for the hint with save mode, I'll try it.

klues, 2020-04-01:
Finally I think I found the problem:
I was starting the keyboard_teleop ROS node at the same time with Webots in a roslaunch-file. The keyboard_teleop ROS node however consumes that much CPU time that loading the Webots world failed.
One thing that remains unclear for me: I also encountered the problem if I started Webots manually, but maybe the ROS node was still running in the background.

I've opened [this PR](https://github.com/cyberbotics/webots_ros/pull/15/files) that should fix the issue. Now both starting with roslaunch-file and starting manually works for me without errors 😃 👍

closed 2020-04-01 by klues