cyberbotics/webots #1071: Variables declared after while (robot.step(timestep) != -1) behave differently (Python)

PrasadNR, 2019-11-07:
**Describe the Bug**
In `mavic2proController.py` file (attached) , I have one line, line 58 `target_y = float(params["yPositionFactor"]) * target_y` before `while (robot.step(timestep) != -1):`. My main objective is in the line 78 `pitch_input = k_pitch_p * pitch - pitch_acceleration + pitchPID(y - target_y)`.

However, if I rewrite this as (for example):

> pitch_input = k_pitch_p * pitch - pitch_acceleration + pitchPID(y - float(params["yPositionFactor"]) * target_y)

It does not work (position hold locks at some different location). Also, the following forms do not work (it is not even consistent with the format of `target_x`).

> while (robot.step(timestep) != -1):
>     target_y1 = float(params["yPositionFactor"]) * target_y
>     roll_input = k_roll_p * roll + roll_acceleration + rollPID(float(params["xPositionFactor"]) * target_x - x)
>     pitch_input = k_pitch_p * pitch - pitch_acceleration + pitchPID(y - target_y1)

**Steps to Reproduce**
1. Install Webots, Python etc. See installation instructions here: https://github.com/PrasadNR/Webots-Quadcopter-Python-SITL#installation-instructions
2. Navigate to "Quadcopter\worlds\Quadcopter.wbt"
3. Now, try changing the codes to the ones I mentioned above.
4. See error

**Expected behavior**
1. Behavior of variables should be consistent.
2. Position of the variable in the variable relative to the while loop shouldn't matter.

**System**
 - Operating System: Windows 8.1
 - Graphics Card: NVIDIA GeForce GTX 860M

**Additional context**
I have attached the files of the project with which I am testing here. I was testing locally before merging this with https://github.com/PrasadNR/Webots-Quadcopter-Python-SITL
[Quadcopter.zip](https://github.com/cyberbotics/webots/files/3818184/Quadcopter.zip)

fabienrohrer, 2019-11-07:
Hi,

This issue is more about Python programming than a Webots issue. This is definitively not the ideal place to ask such questions. We expect to track Webots bugs and feature request here.
=> please give a try to [stack overflow (with the `Webots` tag)](https://stackoverflow.com/questions/tagged/webots) for such kind of questions or on our [Discord chanel](https://discordapp.com/invite/nTWbN9m).

I looked at your issue, and it's because your params have the string format rather than the float format. You should change line 16 to:

```python
		params[line[0]] = float(line[1])
```

closed 2019-11-07 by fabienrohrer

PrasadNR, 2019-11-07:
@fabienrohrer I am converting to `float` (some to `int` and some to `float` depending on the parameters in CSV). Also, this is causing problems even when I do not use CSV and use the parameters directly. Anyways, I can discuss it on discord. Thanks!