cyberbotics/webots #1087: No Lidar Example Code

nicseve, 2019-11-11:
**Documentation page(s) concerned**
https://github.com/cyberbotics/webots/tree/master/projects/samples/devices/controllers/lidar

**What is the problem with the current documentation?**
There is not a single example of the lidar actually being used aside from boiler plate code enabling the point cloud. 

**Describe how you would fix it**
Show at least one example of the lidar features (range image or point cloud) being used to detect objects. At the very least, show how to call the numerous functions associated with the device and display the data they produce.

fabienrohrer, 2019-11-12:
This example uses a lidar to avoid collisions: https://cyberbotics.com/doc/guide/pioneer-3at#pioneer3at-wbt

However, it does not use the point cloud, indeed.

fabienrohrer, 2019-11-20:
I just added an example in #1110.

@nicseve To test it, please install the nightly build of the develop branch of tomorrow, and open `projects/samples/devices/worlds/sick_point_cloud.wbt`

closed 2019-11-20 by fabienrohrer

fabienrohrer, 2019-11-21:
The nightly build containing the sample: https://github.com/cyberbotics/webots/releases/tag/nightly_20_11_2019
The documentation page: https://cyberbotics.com/doc/guide/samples-devices?version=develop#sick_point_cloud-wbt