cyberbotics/webots #1104: Follow up Bug #975 Webot 2019b Rv1 crash when starts at windows 10 pro the Nvidia graphic card

fgw, 2019-11-14:
**Describe the Bug**
Webot 2019b Rv1 suddenly exit during loading nodes at 99%. This issue happens when the system is using the Nvidia graphics card.  When I used the Intel integrate graphics card, the Webot can start normally. 

**Steps to Reproduce**
1. Go to 'Nvidia control panel'
2. Configure the Webots using Nvidia graphics card
3. Start the Webots and wait for the issue happen 

**Expected behavior**
The Webots can work normally with the Nvidia graphics card.

**Screenshots**
Attached videos
Error happened with Nvidia graphics card:
https://drive.google.com/file/d/1377YJ_-86TF3Ymf97vau1y5JUWcVEYdC/view?usp=sharing

Normal with Intel graphics card:
https://drive.google.com/file/d/1WRDAatj1NBaKaIrpk6TX78sFEB2yAzB3/view?usp=sharing

**System**
 - Operating System: windows 10 pro 64 bit 10.0.18362 (simulate windows 7 for Webots)
 - Graphics Card: NVIDIA GeForce 940M 2GB (Driver 441.20)
                            Intel graphics card HD graphics 520  (Driver 21.20.16.4821)

**Additional context**
Add any other context about the problem here.

fabienrohrer, 2019-11-18:
Webots sets up different rendering parameters if you are using an intel or an Nvidia GPU. The expectations on the Intel GPU are lower.

The bug you experiment may come from an OpenGL feature which doesn't work as expected on NVIDIA. To validate this hypothesis, it would be great if you could start Webots on the NVIDIA GPU in safe mode:
https://cyberbotics.com/doc/guide/starting-webots#safe-mode
.. and report here the result.

fabienrohrer, 2019-11-18:
Duplicated with https://github.com/cyberbotics/webots/issues/975 Could we go on the discussion there?

closed 2019-11-18 by fabienrohrer