cyberbotics/webots #1159: Add new Travis and AppVeyor tests

stefaniapedrazzi, 2019-12-09:
Add additional AppVeyor and Travis tests that duplicate the nightly tests so that they can be safely used to test a custom branch without affecting the default PR/branch tests and nightly tests.

Here are some ideas:
- Travis:
  - add a new test (like travis_distrib_no_release) to test the compilation and package creation without creating the draft release page.

- AppVeyor:
 - add a new test that can be used to run a sources test on a custom branch.
 - add a new test (like webots_distrib) that can be used to run a full test like the nightly one on a custom branch.

DavidMansolino, 2020-01-07:
Partially fixed in #1239 (Travis part).

stefaniapedrazzi, 2020-02-17:
From the wiki page (https://github.com/cyberbotics/webots/wiki/Continuous-Integration) it seems that when running the `travis_release_only` the pre-release release on GitHub should not be created and the packages should not be uploaded, is it correct?
If yes, then it doesn't work because the package is still uploaded on GitHub release lists:
- https://travis-ci.com/cyberbotics/webots/builds/149235591
- https://github.com/cyberbotics/webots/releases/tag/nightly_17_2_2020

DavidMansolino, 2020-02-17:
> is it correct?

Yes it is.
I quickly checked the code and it seems correct:
https://github.com/cyberbotics/webots/blob/bb8fd510fa057a3e6d7eab1242cd6e18530fcaf7/.travis.yml#L255
I am suspecting that `TRAVIS_COMMIT_MESSAGE` is not set when starting manually the test from the Travis website. Can you confirm that you did start the test from their website?

stefaniapedrazzi, 2020-02-17:
Yes, I started it from the Travis website

stefaniapedrazzi, 2020-11-30:
Outdated issue: we are now using GitHub actions instead of Travis.

closed 2020-11-30 by stefaniapedrazzi