cyberbotics/webots #1164: Quality Assurance R2020a

DavidMansolino, 2019-12-09:
| Test | Description |  Tested | 
|---|---|---|
| Installation  | Install Webots from the provided package, following the installation instruction from the Webots user guide. | <ul><li>[ ] Windows</li><li>[ ] macOS</li><li>[ ] Snap</li><li>[ ] Tarball</li><li>[x] Debian</li></ul> |

To be continued...

DavidMansolino, 2019-12-09:
Here is a first draft of the table (I only restored what was in our previous quality assurance test without any update or correction).
Maybe we can but this in an issue template so that we can keep it up to date and review it what do you think?

fabienrohrer, 2019-12-09:
This seems good to me as-is.

stefaniapedrazzi, 2019-12-09:
Maybe I would create a wiki page with the template.
I'm not sure we need an issue for this.

stefaniapedrazzi, 2019-12-10:
For the linux tests I would make this:

* all the test for the snap package
* debian and tarball
  - Installation
  - guided tour
  - java/python/cpp controllers
  - movies
  - robot windows
 -  binary compilation

DavidMansolino, 2019-12-10:
I updated the list taking this and the platform seetings from the previous QA into account.

stefaniapedrazzi, 2019-12-11:
We forgot that on Linux we have to test both on Ubuntu 16.04 and Ubuntu 18.04.
I can update the table and add different check boxes for the two versions of Ubuntu.

stefaniapedrazzi, 2019-12-11:
I was wondering if it wasn't better to create a column for each platform.
What do you think?

| Test    | Description | Windows | macOS  | Ubuntu 16.04  | Ubuntu 18.04
|-|-|-|-|-|-
| Installation  | Install Webots from the provided package, following the installation instruction from the Webots user guide. | <ul><li>[ ] tested</li></ul> | <ul><li>[ ] tested</li></ul> | <ul><li>[ ] Snap</li><li>[ ] Tarball</li><li>[ ] Debian</li></ul> | <ul><li>[ ] Snap</li><li>[ ] Tarball</li><li>[ ] Debian</li></ul> |
| Simulation reset  | Check that when pressing the simulation reset while the simulation is running you have the same behavior at the second iteration.Check this for the 10-15 first worlds of the guided tour. | <ul><li>[ ] tested</li></ul> | <ul><li>[ ] tested</li></ul> |  | <ul><li>[ ] Snap</li></ul> |
| Creation of a small project | From scratch and using only the Webots GUI (wizards (including VisualStudio on windows), scene tree, new, and so on), create a world  | <ul><li>[ ] tested</li></ul> | <ul><li>[ ] tested</li></ul> |  | <ul><li>[ ] Snap</li></ul> |

| Test    | Description | Tested Platforms
|-|-|-
| Installation  | Install Webots from the provided package, following the installation instruction from the Webots user guide. | <ul><li>[ ] Windows</li><li>[ ] macOS</li><li>Ubuntu 16.04<ul><li>[ ] Snap</li><li>[ ] Tarball</li><li>[ ] Debian</li></ul><li>Ubuntu 18.04<ul><li>[ ] Snap</li><li>[ ] Tarball</li><li>[ ] Debian</li></ul></ul>
| Simulation reset  | Check that when pressing the simulation reset while the simulation is running you have the same behavior at the second iteration.Check this for the 10-15 first worlds of the guided tour. | <ul><li>[ ] Windows</li><li>[ ] macOS</li><li>Ubuntu 18.04<ul><li>[ ] Snap</li></li></ul></ul>
| Creation of a small project | From scratch and using only the Webots GUI (wizards (including VisualStudio on windows), scene tree, new, and so on), create a world | <ul><li>[ ] Windows</li><li>[ ] macOS</li><li>Ubuntu 18.04<ul><li>[ ] Snap</li></li></ul></ul>

fabienrohrer, 2019-12-11:
My tests on `macOS` went very well.

DavidMansolino, 2019-12-11:
On windows too for me

fabienrohrer, 2019-12-12:
Tests on Ubuntu 16 (Debian and tarball) went well too.

michou214, 2019-12-17:
I also try all the tests on my `Windows` computer. I found minor issues.
1) Tests not tested:
Support UTF-8 characters, Matlab controllers, e-puck remote control, Binary compilation, DARwIn-OP, Streaming server, Extern controllers, SUMO interface, OSM importer and SUMO exporter, ROS controllers

2) `projects/samples/devices/spherical_camera.wbt`: The camera view is not a sphere as you can see in the following screenshot but it's possible that it's due to my graphic card.
[image]
**Configuration OpenGL:**
System: Windows 10 64-bit
OpenGL vendor : Intel
OpenGL renderer : Intel(R) HD Graphics 620
OpenGL version : 3.3.0 - Build 23.20.16.4973
Available GPU memory : N/A

3) `projects/samples/devices/nao_demo.wbt`: I will make a PR to correct the text displayed in the console that doesn't correspond to the movement realized by the Nao. Instead of making one step forward/backward, he makes more.

DavidMansolino, 2019-12-17:
@michou214 thank you for the tests!
I tested on Windows on my computer and ``spherical_camera.wbt`` is working (so we probably have an issue with your GPU).
And about Matlab, I was suspecting that this was broken since a while, but it is good to have the confirmation.

DavidMansolino, 2019-12-17:
Adding a ``.vcxproj.user`` file with this content fixes the problem: 
```
<?xml version="1.0" encoding="utf-8"?>
<Project ToolsVersion="Current" xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Release|x64'">
    <LocalDebuggerEnvironment>PATH=%WEBOTS_HOME%/lib/controller;%PATH%</LocalDebuggerEnvironment>
    <DebuggerFlavor>WindowsLocalDebugger</DebuggerFlavor>
  </PropertyGroup>
</Project>
```

closed 2019-12-19 by stefaniapedrazzi