cyberbotics/webots #1191: Weird behaviour in webots.min.js

michou214, 2019-12-13:
**Describe the Bug**
The `webots.min.js` window that allows you to view a robot in its documentation page has some bugs in the rotation axes (the linear one tested were working fine). In addition, the button to reset the view and axes breaks all the robot configuration.
TIAGo base : wheels
TIAGo Iron : wheels, head and neck
TIAGo Steel : wheels, head, neck and arm
TIAGo Titanium: wheels, head, neck, arm, fingers
TIAGo++ : wheels, head, neck, arms, fingers

**Steps to Reproduce**
_Reset problem_
1. Go to 'https://cyberbotics.com/doc/guide/tiagopp?version=develop'
2. Click on `reset button`

_Rotational axis problem_
1. Go to 'https://cyberbotics.com/doc/guide/tiagopp?version=develop'
2. Move any `RotationalMotor` to see anormal behaviour

**Expected behavior**
The robot should have the same behaviour as in his world example that can be found at `WEBOTS_HOME/projects/robots/pal_robotics/tiago++/protos/Tiago++.proto`

**Screenshots**
_Reset problem_
Before clicking the button
[image]

After clicking the button
[image]

_Axis problem_
[image]

**System**
 - Operating System: Linux Ubuntu 18.04.03 LTS
 - Graphics Card: GeForce GTX 1060 3GB/PCIe/SSE2

**Additional context**
I used the script provided to generate the files used for `webots.min.js`

DavidMansolino, 2019-12-13:
We should either fix this for the release either fallbck to the image temporarily as this gives a very bad user experience.

DavidMansolino, 2019-12-13:
Something is probably broken either on the generation side either on the web side (but for sure not in the robot model). I checked on a simple differential wheel robot and the problem is present too: https://www.cyberbotics.com/doc/guide/turtlebot3-burger?version=feature-turtelbot3

fabienrohrer, 2019-12-16:
Thank you for this report, I will check what is happening this morning.

fabienrohrer, 2019-12-16:
👀

fabienrohrer, 2019-12-16:
This comes from the fact that the Tiago and the TutleBot have rotated solids in their hinges. It's not the case from most of the other robots. This new case is simply not dealt in the robot components.

fabienrohrer, 2019-12-16:
Fixed in #1201

PR #1201 (2019-12-16): Fix robot component rotation
- [x] Fix #1191 Deal with more cases, visible in the Tiago and the TurtleBot, where there is a `Joint.endPoint[Solid].rotation` not set to origin.
- [x] Remove the useless `initialPosition` field in the meta.json files.

Preview: https://cyberbotics.com/doc/guide/tiagopp?version=fix-robot-component-rotation
closed 2019-12-16 by fabienrohrer