cyberbotics/webots #1195: Consistency in PROTO field names for physics

stefaniapedrazzi, 2019-12-16:
It seems that most of the time the field name `enablePhysics` is used, but there are a couple of PROTO nodes that uses `physics` instead.
If possible it would be better to always use the same field name.

Here are the few sensors using `physics`:
* Robotis LDS-01
* SickLdMrs
* MultiSenseS21

Given that it is new, maybe it would be good to fix at least the Robotis LDS-01.

DavidMansolino, 2019-12-16:
I am fixing the Robotis LDS-01 rigth now, and I think we can safetly change the other too @fabienrohrer  will this break anything for our friends using these sensors?

fabienrohrer, 2019-12-16:
I don't see any issue to fix this now, thank you.

DavidMansolino, 2019-12-16:
Fixed in #1196

PR #1196 (2019-12-16): Rename 'physics' fields into 0enablePhysics'
Fix #1195
closed 2019-12-16 by DavidMansolino