cyberbotics/webots #1203: sumo not working for Highway Driving examples on Mac OS 10.3.6 with Macports

WolfgangFahl, 2019-12-16:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1.  Start Highway driving example

**Expected behavior**
sumo should start

**Console output**
Could not connect to TraCI server at localhost:1833 [Errno 61] Connection refused

./sumo --version
dyld: Library not loaded: /opt/X11/lib/libX11.6.dylib
  Referenced from: /Applications/Webots.app/projects/default/resources/sumo/bin/./sumo
  Reason: image not found
Abort trap: 6

**System**
 - Operating System: Mac OS 10.13.6 High Sierra
 - Graphics Card: Radeon RX 560 4096 MB

**Additional context**
XQuartz 1.18.4 (xorg-server 1.18.4)

WolfgangFahl, 2019-12-16:
I am trying:

```
sudo port selfupdate
sudo port -v install xorg
```
according to https://www.xquartz.org/releases/

fabienrohrer, 2019-12-16:
Could you uninstall xorg from port, and install it from the dmg? https://www.xquartz.org

This is the recommended way from our doc: https://cyberbotics.com/doc/automobile/sumo-interface

WolfgangFahl, 2019-12-16:
I am rather happy with my macports xquartz which is now upgraded to 1.20.6 as part of macports 2.6.2.

The needed  dylib is at

```
find /opt/local/lib -name libX11.6.dylib 
/opt/local/lib/libX11.6.dylib
```
and somehow there seems to be hard coded library path /opt/X11/lib in the sumo version you are shipping.

So
```
sudo mkdir /opt/X11
sudo ln -s /opt/local/lib /opt/X11/lib
```
fakes the availability and:

./projects/default/resources/sumo/bin/sumo --version
SUMO Version 0.30.0

works.

WolfgangFahl, 2019-12-16:
Looks good now ..

[image]

fabienrohrer, 2019-12-16:
Great! Well spotted.

WolfgangFahl, 2019-12-16:
You might want to allow specifying the path to sumo as you do allow specifying the path to python. There is a newer sumo version available directly with macports ....

fabienrohrer, 2019-12-16:
This is a good idea. We should upgrade SUMO on every OS too.

DavidMansolino, 2019-12-17:
Updating at each release is costly for us (because we have to make a new package of SUMO to be  embed in Webots every 6 months).

However, what can easily be done which will make the interface way more scalable would be that if `SUMO_HOME` is defined (which is the case when you install SUMO manually) it will use the version of SUMO pointed by this environment variable.

DavidMansolino, 2020-01-08:
The proposed solution has been implemented in #1233.
This will be included in the next version of Webots :smiley:

closed 2020-01-08 by DavidMansolino