cyberbotics/webots #1226: assertion failed: tag < robot.n_device

HefnySco, 2019-12-21:
When I run any of simulations that was running well on previous versions controller exists and gives me:
assertion failed: tag < robot.n_device

**Steps to Reproduce**
1. Open project built on previous version.
2. Run simulator.
 

**System**
 - Ubuntu 18.04
- Webots version 2020a

HefnySco, 2019-12-21:
I discovered the following line fails:

 ```
  WbNodeRef root, node;
  WbFieldRef children, field;
  root = wb_supervisor_node_get_root();
  children = wb_supervisor_node_get_field(root, "children"); // <<< THIS ONE FAILS
```

still dont know why? kindly adivse

megatol, 2019-12-21:
You should probably recompile your controllers with Webots R2020a. Did you try that?

HefnySco, 2019-12-21:
yes sure I did.

it sometimes crashes when calling 
`  children = wb_supervisor_node_get_field(root, "children");
`
I have to reload the simulator not oly reset it to pass the first call.
but even though it failes in the following statement:
and some times crashes at:
`      while (wb_robot_step(timestep) != -1) 
`
then however I keep reset the simulator it does not pass:
`children = wb_supervisor_node_get_field(root, "children");`

HefnySco, 2019-12-21:
What does **

> abort: api/robot.c:347: assertion failed: tag < robot.n_device

** means technically ?

HefnySco, 2019-12-21:
I had to go back to version 2019.

controllers and world exist [here](https://github.com/ArduPilot/ardupilot/tree/master/libraries/SITL/examples/Webots) 
I hope someone can help me to solve this issue.

omichel, 2019-12-22:
Are you sure that you don't have an old `libController.so` library from a previous release that might be in your `PATH` and hence used instead of the one shipped with Webots R2020a?

HefnySco, 2019-12-23:
Maybe. I will make a feash installation on anew machine and let you know.

HefnySco, 2019-12-23:
YES ... It worked seamlessly on fresh installation... 
Thank you

closed 2019-12-23 by HefnySco