cyberbotics/webots #1235: Broken Robot Component

DavidMansolino, 2020-01-03:
**Describe the Bug**
A user reported on Discord that the robot component is broken.
I can indeed reproduce it on Ubuntu 16.04 with firefox.

**Steps to Reproduce**
1. Go to https://cyberbotics.com/doc/guide/aibo-ers7 for example.

**Additional context**
The JS console gives me this error:
```
ReferenceError: webots is not definedviewer.js:890:22
    createRobotComponent viewer.js:890
    populateViewDiv viewer.js:616
    jQuery 4
```

DavidMansolino, 2020-01-06:
Reported by users on Firefox and Chrome on Ubuntu 16.04 and Windows 10 too.

fabienrohrer, 2020-01-06:
The `cyberbotics.com/wwi/R2020a/*` files are corrupted.

fabienrohrer, 2020-01-06:
I just fixed this by manually upload the files from the GitHub repo to the Infomaniak FTP. It seems to work now. I don't understand at which point these files have been corrupted, it was working when we released.

closed 2020-01-06 by fabienrohrer

DavidMansolino, 2020-01-06:
Thank you, I confirm it is now fixed!